project-chip/connectedhomeip #16464: MAC CI compiles gcc even though it claims CLANG only

andy31415, 2022-03-18:
#### Problem
 CI step for clang compiles 5K files instead of linux 1.5K.
 Looking at the output on my machine I see all of:
    - mac_x64_clang
    - mac_x64_gcc
    - mac_x64_gcc_mbedtls
    
 It looks to me like the intent is to compile clang only ... or if that is not the intent the argument should not be given at all and just let some "compile everything" take place.
 
 #### Proposed Solution
 
 Figure out how to really actually only do clang and not the rest. Likely some auto-selection for unified build takes place.

andy31415, 2022-03-18:
We could also just `ninja -C` on the clang directory and also run tests only there to save some time.

andy31415, 2022-03-18:
> We could also just `ninja -C` on the clang directory and also run tests only there to save some time.

That does not seem to work - ninja seems to be top level only.

andy31415, 2022-03-18:
I am trying to retrace steps here and I don't seem to get 5K files anymore, however I see 5.5K files compiled in CI.

andy31415, 2022-03-18:
It seems our CI says `target_os="all"` hence the full build.

andy31415, 2022-03-18:
I tried to split ios and mac instead of "all" in #16325.

andy31415, 2022-03-18:
I believe it is still a build system bug that asking for all os builds triggers an ignore in clang/gcc, however at least that fixes CI

andy31415, 2022-03-19:
It seems that running 

```
scripts/build/gn_gen.sh --args="is_clang=true target_os=\"ios\"" --export-compile-commands
```

for me results in g++ being used to compile stuff instead of clang.

bzbarsky-apple, 2022-03-19:
There is no gcc/g++ on Darwin normally:
```
% gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.29.30)
Target: x86_64-apple-darwin21.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

% g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.29.30)
Target: x86_64-apple-darwin21.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

andy31415, 2022-03-22:
I also wonder why the ios toolkits in https://github.com/project-chip/connectedhomeip/blob/master/build/toolchain/ios/BUILD.gn say they are 'gcc'.

andy31415, 2022-03-22:
Based on additional conversations, it seems `gcc` is actually "Clang from xcode" while `clang` is the pigweed/fuchsia clang. So they differ in version.

It does sound to me as if we probably just want `darwin gcc` to be validated in CI.

bzbarsky-apple, 2022-03-23:
> It does sound to me as if we probably just want darwin gcc to be validated in CI.

I think I agree, for the most part.  @woody-apple

bzbarsky-apple, 2022-03-23:
@vivien-apple

bzbarsky-apple, 2022-03-23:
I created https://github.com/project-chip/connectedhomeip/pull/16551 that might help things some.

That said, maybe `enable_host_clang_build` and `enable_host_gcc_mbedtls_build` should simply default to false when `host_os == "mac"`?  That would have the benefit of also speeding up `gn_build.sh` for people who use it...

bzbarsky-apple, 2022-03-23:
@mspang thoughts on https://github.com/project-chip/connectedhomeip/issues/16464#issuecomment-1075976576 ?

bzbarsky-apple, 2022-03-23:
Looks like the proposed changes in https://github.com/project-chip/connectedhomeip/pull/16551 cut the time for the Darwin build job from ~80 minutes to ~45.

closed 2022-03-23 by andy31415

mspang, 2022-03-23:
> I created #16551 that might help things some.
> 
> That said, maybe `enable_host_clang_build` and `enable_host_gcc_mbedtls_build` should simply default to false when `host_os == "mac"`? That would have the benefit of also speeding up `gn_build.sh` for people who use it...

I'd agree the gcc/clang distinction is significantly less useful on Mac where these are simply different versions of clang. The bootstrapped clang has a decent chance to be more up to date than Xcode's, though, so if you disable it you may not see bleeding edge compiler features or diagnostics until submitting the PR to CI.

The mbedtls question is not as crisp since there is no underlying reason to have a difference in coverage between Mac and Linux by default. I suppose we could make it opt in everywhere (although the original principle of gn_build.sh was to try to get lots of local coverage, to minimize surprises once the change goes to CI. If iterating on something particular it's a lot faster to just build a single binary or test).

bzbarsky-apple, 2022-03-24:
@mspang Thanks, that's helpful.  Will think a bit about how we want to handle this...