project-chip/connectedhomeip #16489: Matter specifications (in progress); how to get access?

henriterhofte, 2022-03-21:
How does one get access to the Matter specifications (in pregress)? 

After some searching, it seems that they are / were available via [https://github.com/CHIP-Specifications/connectedhomeip-spec/](https://github.com/CHIP-Specifications/connectedhomeip-spec/), but this gives a 404. The [Matter Documentation Style Guide](https://project-chip.github.io/connectedhomeip-doc/md__home_runner_work_connectedhomeip_connectedhomeip_docs__s_t_y_l_e__g_u_i_d_e.html) suggests there should be a /docs/specs folder, but I could not find this folder anywhere.

For a new, royalty free open standard, it's nice that this repo is available under an Apache 2.0 license, but it would make matters much easier (pun intended) if the specifications - even if not finalized - would be available as well.

Context: at Windesheim University of Applied Sciences in the Netherlands we're creating the [Twomes open souce energy data collection platform](https://github.com/energietransitie/twomes-generic-esp-firmware). Currently, we are considering using Matter as a basis for a new generatiopn of this platform. We have a Bachelor student working on figuring out whether and how Matter could fit into the picture. [Espressif's blog series on Matter](https://blog.espressif.com/matter-38ccf1d60bcd) already help somewhat, but having access to the full specifications would help a lot. 

Currently, we have a cloud-heavy architecture targeted towards research purposes, with our learning algorithms based on inverse grey-box modelling running in the cloud, but ultimately, we can see such learning algorithms shifting more and more towards the edge. The Matter standard, which focuses a lot on standardisation on edge-to-edge device interoperability seems to allow for a smooth transition path. However, lack of access to the specifications hinder our assessments.

bzbarsky-apple, 2022-05-21:
@clapre can you help out here?

daenney, 2022-11-19:
At this point the Matter specs are available from https://csa-iot.org/developer-resource/specifications-download-request/. This issue can be closed I'm guessing.

closed 2023-06-10 by stale[bot]