project-chip/connectedhomeip #16508: TC-GC-1.2 Chip-tool-darwin testing: AttestationRequest, CSRRequest, AddNOC and CommissionComplete attribute were not present to send

Garik098, 2022-03-21:
As per the steps 5, 6, 7 and 8 in TC-GC-1.2, Controller needs to send  AttestationRequest, CSRRequest, AddNOC and CommissionComplete to the DUT but these attributes were not present in chip-tool-darwin to send. 

Below is the screenshot of chip-tool vs chip-tool-darwin (Above list is chip-tool and below list is chip-tool-darwin). Chip-tool too doesn't have AttestationRequest, CSRRequest and AddNOC commands while CommissionComplete is present in chip-tool and is absent in chip-tool-darwin.
[image]

krypton36, 2022-03-23:
For the test [TC-GC-1.2](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#32-tc-gc-21-breadcrumb-basiccommissioninginfo-regulatoryconfig-and-locationcapability-attributes-dut---commissionee) 
I only see basic-commissioning-info missing. Commission complete is present in the output and for the rest that you mentioned, they appear to be under operation credential. See Output below:
```
○ ./chip-tool-darwin operationalcredentials
[1648060714610] [93511:7737721] CHIP: [TOO] Missing command name
Usage:
  ./chip-tool-darwin operationalcredentials command_name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Commands:                                                                           |
  +-------------------------------------------------------------------------------------+
  | * add-noc                                                                           |
  | * add-trusted-root-certificate                                                      |
  | * attestation-request                                                               |
  | * csrrequest                                                                        |
  | * certificate-chain-request                                                         |
  | * remove-fabric                                                                     |
  | * remove-trusted-root-certificate                                                   |
  | * update-fabric-label                                                               |
  | * update-noc                                                                        |
  | * read                                                                              |
  | * subscribe                                                                         |
  +-------------------------------------------------------------------------------------+
```

bzbarsky-apple, 2022-05-21:
This is working correctly.

closed 2022-05-21 by bzbarsky-apple