project-chip/connectedhomeip #16522: Extended discovery timeout is extended on each `DnssdServer::StartServer`

Damian-Nordic, 2022-03-22:
#### Problem
Extended discovery timeout is reset every time `DnssdServer::StartServer` is called (that is, whenever published DNS-SD services may need to be refreshed) while the timeout should only be scheduled when the extended discovery starts.

 #### Proposed Solution
1. Start the extended discovery timeout in `DnssdServer::StartServer` only if it's not yet started.
2. Cancel the extended discovery timeout when `DnssdServer::StartServer` is called and the commissioning window is open.

woody-apple, 2022-05-26:
Spec Review: We believe this is resolved in the linked PR.

bzbarsky-apple, 2022-05-26:
This is NOT fixed.  This issue is a follow-up from the linked PR, that was identified during review of it.

closed 2022-06-28 by bzbarsky-apple