project-chip/connectedhomeip #16547: Secure Pairing Failed due to Timed-out Issue

RudyGOGOGO, 2022-03-23:
**Basic info about my experiment**
> 1)Computer System: Ubuntu 20.04
> 2)Device: M5Stack Fire Kit
> 3)I am trying to reproduce the work in https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32
> 4)I set the Rendezvous Mode as Bypass
> 5)The device can successfully join my home WiFi network and it also creates its own WiFi network due to the AP mode

I am able to flash my device, and I use the following command to pair the device through my computer however I keep facing the Secure Pairing Failed due to Timed-out Issue
`./chip-tool pairing ble-wifi 12344321 MT-F00-FFF1-8001 '' 20202021 3840`
**Questions:**
> 1)MT-F00-FFF1-8001 is the WiFi SSID of my device, where can I configure the password in menuconfig?
> 2)Per the esp32 readme doc, discriminator and setup-pin-code should be configurable, how should I figure them through menuconfig?
> 3)From the M5Stack log, I can tell discriminator is the default value 3840 and I guess it is used to distinguish different devices, however, how should I confirm the setup-pin-code value? Is there any doc or reference for me to understand how this setup-pin-code works in Matter?

**Log from Computer Terminal**
`
[1648011761.776601][435024:435027] CHIP:BLE: New device scanned: 08:3A:F2:43:85:BA
[1648011761.776629][435024:435027] CHIP:BLE: Device discriminator match. Attempting to connect.
[1648011761.782991][435024:435027] CHIP:BLE: Scan complete notification without an active scan.
[1648011762.093904][435024:435027] CHIP:DL: ConnectDevice complete
[1648011762.292819][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292831][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292834][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service0006
[1648011762.292835][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292837][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service0006
[1648011762.292838][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292839][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service0006
[1648011762.292840][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292844][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292845][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292847][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292848][435024:435027] CHIP:DL: Char1 /org/bluez/hci0/dev_08_3A_F2_43_85_BA/service000a
[1648011762.292851][435024:435027] CHIP:DL: New BLE connection 0x7fbf68035ce0, device 08:3A:F2:43:85:BA, path /org/bluez/hci0/dev_08_3A_F2_43_85_BA
[1648011762.292889][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16385
[1648011762.292898][435024:435029] CHIP:IN: BleConnectionComplete: endPoint 0x55f071a0ca00
[1648011762.947938][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011763.173384][435024:435027] CHIP:DL: Indication received, conn = 0x7fbf68035ce0
[1648011763.173562][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1648011763.173610][435024:435029] CHIP:BLE: peripheral chose BTP version 4; central expected between 4 and 4
[1648011763.173624][435024:435029] CHIP:BLE: using BTP fragment sizes rx 244 / tx 244.
[1648011763.173634][435024:435029] CHIP:BLE: local and remote recv window size = 5
[1648011763.173665][435024:435029] CHIP:IN: BLE EndPoint 0x55f071a0ca00 Connection Complete
[1648011763.173977][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16388
[1648011763.174004][435024:435029] CHIP:BLE: subscribe complete, ep = 0x55f071a0ca00
[1648011763.353019][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011765.827649][435024:435027] CHIP:DL: Indication received, conn = 0x7fbf68035ce0
[1648011765.827741][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1648011768.572116][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011771.129897][435024:435027] CHIP:DL: Indication received, conn = 0x7fbf68035ce0
[1648011771.130029][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1648011773.837712][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011776.441971][435024:435027] CHIP:DL: Indication received, conn = 0x7fbf68035ce0
[1648011776.442072][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1648011779.192746][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011781.749965][435024:435027] CHIP:DL: Indication received, conn = 0x7fbf68035ce0
[1648011781.750117][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1648011784.457999][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011786.969871][435024:435027] CHIP:DL: Indication received, conn = 0x7fbf68035ce0
[1648011786.969963][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1648011789.812777][435024:435029] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1648011791.605835][435024:435029] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1648011791.605880][435024:435029] CHIP:TOO: Secure Pairing Failed
[1648011791.605898][435024:435029] CHIP:TOO: Pairing Failure: ../../src/protocols/secure_channel/PASESession.cpp:324: CHIP Error 0x00000032: Timeout
`

**Log from M5Stack Device during the pairing process**
`
I (38866759) chip[DL]: BLE GAP connection established (con 0)
I (38866759) chip[DL]: CHIPoBLE advertising stopped
I (38866769) app-devicecallbacks: Current free heap: 69592
[... 41 lines cut ...]
I (38895849) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0 notify = 0
I (38895849) chip[BLE]: Releasing end point's BLE connection back to application.
I (38895849) chip[DL]: CHIPoBLE unsubscribe received
I (38895869) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MT-3840)
I (38895879) NimBLE: GAP procedure initiated: advertise; 
I (38895879) NimBLE: disc_mode=2
I (38895889) NimBLE:  adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (38895889) NimBLE: 

I (38895909) chip[DL]: CHIPoBLE advertising started
I (38895909) app-devicecallbacks: Current free heap: 69592

I (38898729) chip[DL]: BLE GAP connection terminated (con 0 reason 0x213)
I (38898729) app-devicecallbacks: CHIPoBLE disconnected
I (38898729) app-devicecallbacks: Current free heap: 69592

I (38898739) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MT-3840)
I (38898759) chip[DL]: Device already advertising, stop active advertisement and restart
I (38898759) NimBLE: GAP procedure initiated: stop advertising.

I (38898769) NimBLE: GAP procedure initiated: advertise; 
I (38898769) NimBLE: disc_mode=2
I (38898769) NimBLE:  adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (38898779) NimBLE: 
`

RudyGOGOGO, 2022-03-23:
@woody-apple @bzbarsky-apple @yufengwangca would you mind providing some insight regarding the issue above?

RudyGOGOGO, 2022-03-24:
I also tried the Android Controller according to https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md, I used the "provision the device with WiFi network" and scan the QR code. 
1)The device can be found
2)The pairing still fails with the same device log shown below
3)The Android app does not have the timeout error, looks like it keep trying to pair 

> I (38880799) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0
> I (38883609) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
> I (38886109) [...]

RudyGOGOGO, 2022-03-24:
I also tried another board: ESP32-DevkitC-VE, it has the same result as M5Stack Fire Kit

bzbarsky-apple, 2022-03-24:
Can you please attach (as a file, not a paste, since it will be long) the complete log from the m5stack?

shubhamdp, 2022-03-24:
@RudyGOGOGO, I see you are using rendezvous mode as bypass and trying to commission device using BLE transport. Can you try using BLE rendezvous mode.

bzbarsky-apple, 2022-03-24:
Oh!  Yes, that's not going to work.  The bypass rendezvous stuff does not work at all, and certainly commissioning over BLE if it's set won't work either.

bzbarsky-apple, 2022-03-24:
Note https://github.com/project-chip/connectedhomeip/pull/16597 from earlier today which might at least remove the footgun.

RudyGOGOGO, 2022-03-24:
MUCH APPRECIATED from your responses! You saved my days! It works for Android Controller.
I think it would be better to update the README.MD doc in https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32, I talked to my friend, he also thought Bypass is suggested according to the statement below
> Select the Rendezvous Mode via Demo -> Rendezvous Mode. If Rendezvous Mode is ByPass then set the credentials of the WiFi Network (i.e. SSID and Password from menuconfig).
> idf.py menuconfig -> Component config -> CHIP Device Layer -> WiFi Station Options

Btw, would you mind checking the new issue about ChipTool Command Line
https://github.com/project-chip/connectedhomeip/issues/16600

bzbarsky-apple, 2022-03-24:
> I think it would be better to update the README.MD doc

Good catch.  I will add that to #16597.

closed 2022-03-24 by bzbarsky-apple