project-chip/connectedhomeip #16550: Pump configuration cluster attributes that should be nullable are not marked that way in the XML

manjunathambiger23, 2022-03-23:
Summary: 
              As per spec the default value of some attributes are null and we need to check the default value as expected using Scripts but attributes that should be nullable are not marked that way in the XML

Spec screenshot:
[image]
Execution screenshot:
[image]

manjunathambiger23, 2022-04-05:
@bzbarsky-apple still facing build error 
[image]

bzbarsky-apple, 2022-04-05:
@manjunathambiger23 Sure, it's still not marked nullable.  Please talk to the pump configuration folks....

tlykkeberg-grundfos, 2022-04-07:
Fixed in #17181

PR #17181 (2022-04-20): [XML] Correct the PCC cluster xml to reflect the specification
#### Problem
* The PCC cluster xml does not fully reflect the specification
* Min and max values for some attributes in the PCC was wrong
* Add some non-default access priviledges to the PCC
* Set the ClusterRevision to be '3'
* Fixes #16550
* Fixes #17112
* Fixes #9769

#### Change overview
* Added `isNullable="true"` to the attributes which is specified to have this posibillity
* Corrected the `min` and `max` for all attributes to reflect the specification
* Added RW VM to the LifeTimeRunningHours and LifeTimeConsumedEnergy attri [... cut]