project-chip/connectedhomeip #16552: [TC-ETHDIAG-1.1] attribute with default as "null" showing failure response

manjunathambiger23, 2022-03-23:
Summary:
               As per spec default value of PHYRate attribute is "null" showing failure response when we executed. 
               
Spec screenshot:
[image]

Execution logs and screenshot:
[image]
[ethdiag.txt](https://github.com/project-chip/connectedhomeip/files/8330554/ethdiag.txt)

manjunathambiger23, 2022-04-05:
@bzbarsky-apple still facing this issue if we try to read the null value.
[image]

bzbarsky-apple, 2022-04-05:
That's an UNSUPPORTED_CLUSTER response.  Is the cluster involved actually enabled in the server you are trying to read from @manjunathambiger23 ?

manjunathambiger23, 2022-04-14:
@bzbarsky-apple i uploaded wrong logs i tried it in endpoint 0 its still showing failure response

bzbarsky-apple, 2022-04-18:
@manjunathambiger23 Is the response still UNSUPPORTED_CLUSTER?

Which _exact_ commands are being run on which exact SHA on which exact example apps?

manjunathambiger23, 2022-04-19:
@bzbarsky-apple the response is: expected to be null but isn't 
[image]
[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/8510393/exe_log.txt)

bzbarsky-apple, 2022-04-19:
The spec for PHYRate says:
```
The PHYRate attribute SHALL indicate the current nominal, usable speed at the top of the physical layer of the Node. A value of null SHALL indicate that the interface is not currently configured or operational.
```
It's coming back as "5", which is "Rate10G".

Why are you expecting null?  Is the interface in question in fact "not configured or operational"?  What is the _exact_ server setup?

bzbarsky-apple, 2022-05-21:
Resolving, given correct-looking SDK behavior and no response.

closed 2022-05-21 by bzbarsky-apple