project-chip/connectedhomeip #16554: Wi-Fi Network Diagnostics Cluster attributes with default value "null" showing response as "expected to be null but isn't"

manjunathambiger23, 2022-03-23:
Summary:
              As per spec Wi-Fi Network Diagnostics Cluster some  attributes with default value "null" showing response as "expected to be null but isn't"  when we executed
               
Spec screenshot:
[image]

Execution logs and screenshot:
[image]
[wifidiag.txt](https://github.com/project-chip/connectedhomeip/files/8330938/wifidiag.txt)

bzbarsky-apple, 2022-05-28:
@manjunath-grl @cjandhyala Is this still an issue?

manjunath-grl, 2022-06-09:
@bzbarsky-apple issue is resolved can you please close it.

closed 2022-06-10 by bzbarsky-apple