project-chip/connectedhomeip #16561: ESP build is failing

sowmyassp, 2022-03-23:
Issue : Not able to build ( idf.py build ) for SDK test execution using ESP32 device.

Commit used: ab00e9c231f0763d2b0b82787d44949715244adb

Commands `idf.py build`

Error Log:

> In file included from /home/grl/tools_mar23/esp-idf/components/mbedtls/mbedtls/include/mbedtls/build_info.h:59,
                 from /home/grl/tools_mar23/esp-idf/components/mbedtls/mbedtls/library/common.h:26,
                 from /home/grl/tools_mar23/esp-idf/components/mbedtls/mbedtls/library/x509_create.c:20:
/home/grl/tools_mar23/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h:30:10: fatal error: mbedtls/config.h: No such file or directory
 #include "mbedtls/config.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[12/1267] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
otadata,data,ota,0xf000,8K,
phy_init,data,phy,0x11000,4K,
ota_0,app,ota_0,0x20000,1500K,
ota_1,app,ota_1,0x1a0000,1500K,
ot_storage,data,58,0x317000,8K,
*******************************************************************************
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

shubhamdp, 2022-03-23:
@sowmyassp, can you please tell me which app you are trying to build?

chshu, 2022-03-24:
@sowmyassp Please use the ESP-IDF v4.4 as documented in the README. It's a known issue only with IDF latest master branch, the fix is on the way.

sowmyassp, 2022-03-29:
@shubhamdp all-clusters-app

@chshu - Ok, thank you

sowmyassp, 2022-04-04:
@chshu Still facing same issue with commit - 41a431dcc7c123699818636afd01b57412a6f3a4		

> In file included from /home/grl/tools_apr4/esp-idf/components/mbedtls/mbedtls/include/mbedtls/build_info.h:59,
                 from /home/grl/tools_apr4/esp-idf/components/mbedtls/mbedtls/library/common.h:26,
                 from /home/grl/tools_apr4/esp-idf/components/mbedtls/mbedtls/library/x509_crl.c:30:
/home/grl/tools_apr4/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h:31:10: fatal error: mbedtls/config.h: No such file or directory
 #include "mbedtls/config.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[12/1262] Generating ../../partition_table/partition-table.bin

PSONALl, 2022-04-05:
@sowmyassp can you share the ESP-IDF commit id, you are using?

wqx6, 2022-04-06:
@sowmyassp Please use [ESP-IDF v4.4](https://github.com/espressif/esp-idf/tree/v4.4) for ESP32/ESP32C3. we now only use IDF v5.0-dev(current master branch) for ESP32H2.

dhrishi, 2022-04-13:
Closing the issue assuming it is working fine. Feel free to re-open

closed 2022-04-13 by dhrishi