project-chip/connectedhomeip #16573: [TC-IDM-6.1]SDK support required for the Events Read Interaction from TH to DUT

Rajashreekalmane, 2022-03-23:
#### Problem
  In [TC-IDM-6.1] test case the test step 8,9 and 10 was out of scope for TE-8 and
 doesn't had the verification steps

PFA Test-plan screenshot:-
[image]

Do we have the SDK developed to support these steps? If it is , please share the verification steps for the same.

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#event-information-blocks-and-paths

vivien-apple, 2022-03-28:
@mrjerryjohns Can you provide what is the device configuration that should be used for such test cases ?  It is unclear to me that this is an issue requesting code changes instead of the steps missing informations.

Rajashreekalmane, 2022-04-22:
@vivien-apple  This command is not working, I am getting

https://user-images.githubusercontent.com/95841834/164677732-3f76c421-e67e-4b48-9a07-0f22c42844f5.png

Steps to reproduce:- ./chip-tool basic read-event start-up 0x12344321 0 --commissioner-fabricid 332211

vivien-apple, 2022-04-29:
For the last step you don't need to use the `commissioner-fabricid` thing. Basically if on the DUT exists some events that are fabric scoped you can just use the classic `read-event` function and observe what happens.

For example in the case of the `all-clusters-app` that contains `TestCluster`, you can validate it by following those steps:
```
$ ./out/debug/standalone/chip-tool pairing qrcode 0x12344321 MT:-24J029Q00KA0648G00
$ ./out/debug/standalone/chip-tool administratorcommissioning open-basic-commissioning-window 180 0x12344321 0 --timedInteractionTimeoutMs 1000
$ ./out/debug/standalone/chip-tool pairing qrcode 0x12344321 MT:-24J029Q00KA0648G00 --commissioner-name beta
$ ./out/debug/standalone/chip-tool testcluster test-emit-test-fabric-scoped-event-request 1 0x12344321 1 
$ ./out/debug/standalone/chip-tool testcluster read-event test-fabric-scoped-event 0x12344321 1 --commissioner-name alpha
# You should see the event
$ ./out/debug/standalone/chip-tool testcluster read-event test-fabric-scoped-event 0x12344321 1 --commissioner-name beta
# You should not see the event
```

closed 2022-04-29 by vivien-apple