project-chip/connectedhomeip #16581: [TC-DM-2.3] `operationalcredentials` cluster command failures using chip-tool-darwin

liamgonyea, 2022-03-23:
#### Problem
When running Steps 2, 4, and 5 for `[TC-DM-2.3]`, chip-tool-darwin received an error response of `General error: 0x01` when running the following commands:
Step 2: ./out/debug/chip-tool-darwin operationalcredentials attestation-request 1234 0
Step 4: ./out/debug/chip-tool-darwin operationalcredentials csrrequest 1234 0
Step 5: ./out/debug/chip-tool-darwin operationalcredentials add-trusted-root-certificate 1234 0

#### Testing Setup
TE8 commit: `448578dd09004679d10a7fa883f2e680de05079d`
DUT: chip-tool-darwin
TH: chip-all-clusters-app

 #### Test Case
3.2.3. [TC-DM-2.3] Operational Credential Commands [DUT - Commissioner]
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#323-tc-dm-23-operational-credential-commands-dut---commissioner

[image]

 #### Logs
Failures:
Step 2: Attestation Response

> [1648057907557] [62389:43782744] CHIP: [-] IM Error 0x00000501: General error: 0x01 at ../../commands/common/CHIPCommandBridge.mm:51
> [1648057907557] [62389:43782744] CHIP: [TOO] Run command failure: IM Error 0x00000501: General error: 0x01
> 

Step 4: CSR Response

> [1648057978402] [62406:43783495] CHIP: [-] IM Error 0x00000501: General error: 0x01 at ../../commands/common/CHIPCommandBridge.mm:51
> [1648057978402] [62406:43783495] CHIP: [TOO] Run command failure: IM Error 0x00000501: General error: 0x01

Step 5: Add Trusted Root Certificate Request

> [1648058014037] [62417:43784015] CHIP: [-] IM Error 0x00000501: General error: 0x01 at ../../commands/common/CHIPCommandBridge.mm:51
> [1648058014037] [62417:43784015] CHIP: [TOO] Run command failure: IM Error 0x00000501: General error: 0x01

[consolidated_mac_accessory_TC-DM-2-3.txt](https://github.com/project-chip/connectedhomeip/files/8335805/consolidated_mac_accessory_TC-DM-2-3.txt)

bzbarsky-apple, 2022-05-21:
> Step 2: ./out/debug/chip-tool-darwin operationalcredentials attestation-request 1234 0

Not a valid command; missing attestation nonce.  Doesn't even run on tip.

> Step 4: ./out/debug/chip-tool-darwin operationalcredentials csrrequest 1234 0

Missing CSR nonce.  Does not work on tip.

> Step 5: ./out/debug/chip-tool-darwin operationalcredentials add-trusted-root-certificate 1234 0

Missing the root certificate, no?

Please retest on tip.

bzbarsky-apple, 2022-05-28:
@liamgonyea See above?

@cjandhyala

liamgonyea, 2022-11-30:
Verified that this is no longer an issue after running the above commands with `darwin-framework-tool` on commit: `c279578c5bc37f117335aa96cec6c5552f070cc0`

closed 2022-11-30 by liamgonyea