project-chip/connectedhomeip #16584: TC-IDM-6.4 chip tool Darwin testing: Eventrequests and EventFilters were missing in the response for step 1

Garik098, 2022-03-23:
As per step 1 in TC-IDM-6.4, the response should contain EventRequests and EventFilters, but the response for subscribe were missing these fields. Below is the expected outcome.
[image]

Below is the outcome for the command - ./chip-tool-darwin thermostat subscribe cluster-revision 1 10 1 1 1
[image]

Controller side log:
adk@adks-Mac-mini debug % ./chip-tool-darwin thermostat subscribe cluster-revision 1 10 1 1 1
[1648065573359] [39479:3699021] CHIP: [TOO] Running Command
2022-03-23 12:59:33.363 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: ChipToolOpCredsCAKey, value BAG6TxndA16E328KB+s67FX9+5LHU3I7cBrMiHJcajq1An4MxepBioxfnjaunMfI0vAeiWTeGU8HTgmQJVfHOAVnEv1hZWowEttLnZKxu5hy5AIn9FY0u9QOq48M2kXqmA==
2022-03-23 12:59:33.365 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: com.zigbee.chip.commissioner.device_id
2022-03-23 12:59:33.365 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: com.zigbee.chip.commissioner.device_id, value Gr6DOjXiCDA=
2022-03-23 12:59:33.366 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: MatterCARootCert0
2022-03-23 12:59:33.366 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: MatterCARootCert0, value MIIB2TCCAYCgAwIBAgIBADAKBggqhkjOPQQDAjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwHhcNMjIwMzA4MTUwMDAxWhcNMjMwMzA4MTUwMDAxWjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAAQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFo2MwYTAPBgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBBjAdBgNVHQ4EFgQUpAxXUsDrlQkg7iOO2exDAIyjA7wwHwYDVR0jBBgwFoAUpAxXUsDrlQkg7iOO2exDAIyjA7wwCgYIKoZIzj0EAwIDRwAwRAIgQUDQA6aWAftkPOHso2ZV3++c1BRhHsD1MZ3vK0lauJcCIEZEVgTELLz7k+XFXsETqa4U1jXFK3GF+Zyi2DvqqfmU
[1648065573366] [39479:3699021] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1648065573373] [39479:3699021] CHIP: [DL] _Init
[1648065573373] [39479:3699021] CHIP: [IN] UDP::Init bind&listen port=5541
[1648065573375] [39479:3699021] CHIP: [IN] UDP::Init bound to port=5541
[1648065573375] [39479:3699021] CHIP: [IN] UDP::Init bind&listen port=5541
[1648065573375] [39479:3699021] CHIP: [IN] UDP::Init bound to port=5541
[1648065573375] [39479:3699021] CHIP: [IN] TransportMgr initialized
[1648065573375] [39479:3699021] CHIP: [DIS] Init fabric pairing table with server storage
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F01/Fabric1
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F01/Fabric1, value AQAAAAAAAAAavoM6NeIIMAAAAAAAAAAAAADvAAAA/AAE4DFh8ZQ0tNQWhy2zZhR7hKLLXtBT7c4nqAys8VsPVAmVfY4IWOkwCGvh7S16/4xP1mPfxNJWTSXpsM8kaIYYrwE1VKTVzrgo21ldtxwiPn8R9I81i2JRVt1bLMfrqwYlAAAAAAAAAGEAAAAAAAAAFTABAQAkAgE3AyUU0gQkFQAYJgTxK7opJgVxX5srNwYlFNIEJBUAGCQHASQIATAJQQQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFNwo1ASkBGCQCYDAEFKQMV1LA65UJIO4jjtnsQwCMowO8MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAQUDQA6aWAftkPOHso2ZV3++c1BRhHsD1MZ3vK0lauJdGRFYExCy8+5PlxV7BE6muFNY1xStxhfmcotg76qn5lBgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAVMAEBASQCATcDJRTSBCQVABgmBPEruikmBXFfmys3BiQVACcRGr6DOjXiCDAYJAcBJAgBMAlBBOAxYfGUNLTUFocts2YUe4Siy17QU+3OJ6gMrPFbD1QJlX2OCFjpMAhr4e0tev+MT9Zj38TSVk0l6bDPJGiGGK83CjUBKAEYJAIBNgMEAgQBGDAEFE5MrUA6J9TLEXIDswqDh+XQhN25MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAwaRsIg0BJ16NFDi9Tei62tsnhrhlv/DGr3JxPjTx+MDE1m1JyVObAGfLE4fUlLHz7sNhWEgCBjqEi5ZNR6ZVQRgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
[1648065573375] [39479:3699021] CHIP: [IN] Loading certs from storage
[1648065573375] [39479:3699021] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000000 and root pubkey
[1648065573375] [39479:3699021] CHIP: [IN] 0x04, 0x01, 0xba, 0x4f, 0x19, 0xdd, 0x03, 0x5e, 
[1648065573375] [39479:3699021] CHIP: [IN] 0x84, 0xdf, 0x6f, 0x0a, 0x07, 0xeb, 0x3a, 0xec, 
[1648065573375] [39479:3699021] CHIP: [IN] 0x55, 0xfd, 0xfb, 0x92, 0xc7, 0x53, 0x72, 0x3b, 
[1648065573375] [39479:3699021] CHIP: [IN] 0x70, 0x1a, 0xcc, 0x88, 0x72, 0x5c, 0x6a, 0x3a, 
[1648065573375] [39479:3699021] CHIP: [IN] 0xb5, 0x02, 0x7e, 0x0c, 0xc5, 0xea, 0x41, 0x8a, 
[1648065573375] [39479:3699021] CHIP: [IN] 0x8c, 0x5f, 0x9e, 0x36, 0xae, 0x9c, 0xc7, 0xc8, 
[1648065573375] [39479:3699021] CHIP: [IN] 0xd2, 0xf0, 0x1e, 0x89, 0x64, 0xde, 0x19, 0x4f, 
[1648065573375] [39479:3699021] CHIP: [IN] 0x07, 0x4e, 0x09, 0x90, 0x25, 0x57, 0xc7, 0x38, 
[1648065573375] [39479:3699021] CHIP: [IN] 0x05, 
[1648065573375] [39479:3699021] CHIP: [IN] Generated compressed fabric ID
[1648065573375] [39479:3699021] CHIP: [IN] 0xd0, 0x99, 0xe4, 0x0d, 0x93, 0xb1, 0xdc, 0x0c, 
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F02/Fabric2
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F02/Fabric2, value (null)
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F03/Fabric3
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F03/Fabric3, value (null)
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F04/Fabric4
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F04/Fabric4, value (null)
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F05/Fabric5
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F05/Fabric5, value (null)
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F06/Fabric6
2022-03-23 12:59:33.375 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F06/Fabric6, value (null)
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F07/Fabric7
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F07/Fabric7, value (null)
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F08/Fabric8
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F08/Fabric8, value (null)
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F09/Fabric9
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F09/Fabric9, value (null)
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F0A/Fabrica
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F0A/Fabrica, value (null)
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F0B/Fabricb
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] CHIPPersistentStorageDelegate Get Value for Key: F0B/Fabricb, value (null)
2022-03-23 12:59:33.376 chip-tool-darwin[39479:3699021] PersistentStorageDelegate Sync Get Value for Key: F0C/Fabricc
[... cut ...]
[1648065573.545217][1860:1860] CHIP:DMG: {
[1648065573.545243][1860:1860] CHIP:DMG: 	Status = 0x0,
[1648065573.545272][1860:1860] CHIP:DMG: 	InteractionModelRevision = 1
[1648065573.545298][1860:1860] CHIP:DMG: }
[1648065573.545325][1860:1860] CHIP:IM: Received status response, status is 0
[1648065573.545365][1860:1860] CHIP:DMG: Refresh Subscribe Sync Timer with max 10 seconds
[1648065573.545397][1860:1860] CHIP:DMG: <RE> OnReportConfirm: NumReports = 0
[1648065573.545421][1860:1860] CHIP:DMG: IM RH moving to [GeneratingReports]

krypton36, 2022-04-06:
@Garik098 I don't see those responses in chip tool see log below:
```
[1649281321240] [65075:6225227] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:1189803 on exchange 57810i
[1649281321240] [65075:6225227] CHIP: [EM] Found matching exchange: 57810i, Delegate: 0x6000000a40f0
[1649281321240] [65075:6225227] CHIP: [EM] Rxd Ack; Removing MessageCounter:5804149 from Retrans Table on exchange 57810i
[1649281321240] [65075:6225227] CHIP: [EM] Removed CHIP MessageCounter:5804149 from RetransTable on exchange 57810i
[1649281321240] [65075:6225227] CHIP: [DMG] ReportDataMessage =
[1649281321240] [65075:6225227] CHIP: [DMG] {
[1649281321240] [65075:6225227] CHIP: [DMG] 	SubscriptionId = 0xd5a3da6b6165b35a,
[1649281321240] [65075:6225227] CHIP: [DMG] 	AttributeReportIBs =
[1649281321240] [65075:6225227] CHIP: [DMG] 	[
[1649281321240] [65075:6225227] CHIP: [DMG] 		AttributeReportIB =
[1649281321240] [65075:6225227] CHIP: [DMG] 		{
[1649281321240] [65075:6225227] CHIP: [DMG] 			AttributeDataIB =
[1649281321240] [65075:6225227] CHIP: [DMG] 			{
[1649281321240] [65075:6225227] CHIP: [DMG] 				DataVersion = 0x365579c0,
[1649281321240] [65075:6225227] CHIP: [DMG] 				AttributePathIB =
[1649281321240] [65075:6225227] CHIP: [DMG] 				{
[1649281321240] [65075:6225227] CHIP: [DMG] 					Endpoint = 0x1,
[1649281321240] [65075:6225227] CHIP: [DMG] 					Cluster = 0x201,
[1649281321240] [65075:6225227] CHIP: [DMG] 					Attribute = 0x0000_FFFD,
[1649281321240] [65075:6225227] CHIP: [DMG] 				}
[1649281321240] [65075:6225227] CHIP: [DMG] 					
[1649281321240] [65075:6225227] CHIP: [DMG] 					Data = 3, 
[1649281321240] [65075:6225227] CHIP: [DMG] 			},
[1649281321240] [65075:6225227] CHIP: [DMG] 			
[1649281321240] [65075:6225227] CHIP: [DMG] 		},
[1649281321240] [65075:6225227] CHIP: [DMG] 		
[1649281321240] [65075:6225227] CHIP: [DMG] 	],
[1649281321240] [65075:6225227] CHIP: [DMG] 	
[1649281321240] [65075:6225227] CHIP: [DMG] 	InteractionModelRevision = 1
[1649281321240] [65075:6225227] CHIP: [DMG] }
[1649281321241] [65075:6225227] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_FFFD DataVersion: 911571392
[1649281321242] [65075:6225227] CHIP: [TOO]   ClusterRevision: 3

```

Garik098, 2022-04-06:
After discussion on getting more clarity on the test step and the way to perform it, closing this ticket and opening new ticket https://github.com/project-chip/connectedhomeip/issues/17151

closed 2022-04-06 by Garik098