project-chip/connectedhomeip #16600: With a40104a commit, ChipTool Command without paa-trust-store-path cannot work

RudyGOGOGO, 2022-03-24:
This afternoon, the following command can build the connection at least, even though there is timeout issue(please check this link: https://github.com/project-chip/connectedhomeip/issues/16547)
`./chip-tool pairing ble-wifi 12344321 MyHomeWiFiSSID MyHomeWiFiPassword 20202021 3840`
However, with the commit, the same command triggers the following error
[ChipToolCommandLogs.txt](https://github.com/project-chip/connectedhomeip/files/8338869/ChipToolCommandLogs.txt)

bzbarsky-apple, 2022-03-24:
Looks like this is erroring out in the code added in #16125.  @vijs could you please take a look?

bzbarsky-apple, 2022-03-24:
In particular, the `"."` default path for chip-tool if `--paa-trust-store-path` is not specified doesn't actually work, because there are no `.der` files in there.

closed 2022-03-24 by vijs