project-chip/connectedhomeip #16614: [TC-LO-1.1]- While passing SupportedLocales attribute to write we will get 0x87 instead 0x88.

Rajashreekalmane, 2022-03-24:
#### Problem
If we don't have Access to write command for SupportedLocales attribute in spec, if we try to write we should get error code as UNSUPPORTED_WRITE(0x88) but currently in chip-tool responds  CONSTRAINT_ERROR(0x87)

 * expected behavior:-UNSUPPORTED_WRITE(0x88)
 * actual behavior:- CONSTRAINT_ERROR(0x87)

 * steps to reproduce:- sudo ./chip-tool any write-by-id 0x002B 1 newsupportedlocales 3 0

 *PFA screenshot below:-

grl@grl-Inspiron-7572:~/Mar21_cntrl/connectedhomeip/examples/chip-tool/out/debug$ sudo ./chip-tool any write-by-id 0x002B 1 newsupportedlocales 3 0
[1648121158.319584][13467:13467] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1648121158.319701][13467:13467] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1648121158.319730][13467:13467] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1648121158.319749][13467:13467] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1648121158.319825][13467:13467] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-F9a6gv)
[1648121158.319964][13467:13467] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648121158.319975][13467:13467] CHIP:DL: NVS set: chip-counters/reboot-count = 10 (0xA)
[1648121158.320921][13467:13467] CHIP:DL: Got Ethernet interface: enp2s0
[1648121158.321389][13467:13467] CHIP:DL: Found the primary Ethernet interface:enp2s0
[1648121158.321843][13467:13467] CHIP:DL: Got WiFi interface: wlp3s0
[1648121158.323614][13467:13467] CHIP:DL: Found the primary WiFi interface:wlp3s0
[1648121158.323644][13467:13467] CHIP:IN: UDP::Init bind&listen port=5542
[1648121158.323668][13467:13467] CHIP:IN: UDP::Init bound to port=5542
[1648121158.323672][13467:13467] CHIP:IN: BLEBase::Init - setting/overriding transport
[1648121158.323677][13467:13467] CHIP:IN: TransportMgr initialized
[1648121158.323688][13467:13467] CHIP:DIS: Init fabric pairing table with server storage
[1648121158.323693][13467:13467] CHIP:IN: Loading from storage for fabric index 1
[1648121158.323828][13467:13467] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1648121158.323836][13467:13467] CHIP:IN: 0x04, 0xe5, 0xbe, 0x64, 0xbe, 0x2b, 0x1c, 0x97, 
[1648121158.323840][13467:13467] CHIP:IN: 0x76, 0x89, 0xfc, 0xfe, 0xd8, 0x7b, 0x56, 0x4b, 
[1648121158.323844][13467:13467] CHIP:IN: 0x05, 0x9a, 0x89, 0x18, 0xf3, 0x32, 0x4b, 0x70, 
[1648121158.323847][13467:13467] CHIP:IN: 0xee, 0x90, 0x95, 0x4b, 0x89, 0xb0, 0x0f, 0x0e, 
[1648121158.323851][13467:13467] CHIP:IN: 0x9f, 0x03, 0x2a, 0x5f, 0xf5, 0x25, 0xdd, 0xcc, 
[1648121158.323855][13467:13467] CHIP:IN: 0xae, 0xaf, 0x5e, 0x03, 0x26, 0xfd, 0x31, 0x2f, 
[1648121158.323858][13467:13467] CHIP:IN: 0xa4, 0x0a, 0xfc, 0xda, 0x30, 0x2a, 0x0e, 0xc5, 
[1648121158.323862][13467:13467] CHIP:IN: 0x56, 0x9a, 0x7a, 0xfd, 0xe1, 0x0f, 0xc8, 0x5c, 
[1648121158.323865][13467:13467] CHIP:IN: 0xa8, 
[1648121158.323890][13467:13467] CHIP:IN: Generated compressed fabric ID
[1648121158.323895][13467:13467] CHIP:IN: 0x3b, 0x08, 0xaf, 0xc9, 0xa9, 0xaf, 0xfc, 0x74, 
[1648121158.324757][13467:13467] CHIP:IN: Loading from storage for fabric index 2
[1648121158.324855][13467:13467] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1648121158.324862][13467:13467] CHIP:IN: 0x04, 0xe5, 0xbe, 0x64, 0xbe, 0x2b, 0x1c, 0x97, 
[1648121158.324866][13467:13467] CHIP:IN: 0x76, 0x89, 0xfc, 0xfe, 0xd8, 0x7b, 0x56, 0x4b, 
[1648121158.324869][13467:13467] CHIP:IN: 0x05, 0x9a, 0x89, 0x18, 0xf3, 0x32, 0x4b, 0x70, 
[1648121158.324873][13467:13467] CHIP:IN: 0xee, 0x90, 0x95, 0x4b, 0x89, 0xb0, 0x0f, 0x0e, 
[1648121158.324877][13467:13467] CHIP:IN: 0x9f, 0x03, 0x2a, 0x5f, 0xf5, 0x25, 0xdd, 0xcc, 
[1648121158.324880][13467:13467] CHIP:IN: 0xae, 0xaf, 0x5e, 0x03, 0x26, 0xfd, 0x31, 0x2f, 
[1648121158.324884][13467:13467] CHIP:IN: 0xa4, 0x0a, 0xfc, 0xda, 0x30, 0x2a, 0x0e, 0xc5, 
[1648121158.324888][13467:13467] CHIP:IN: 0x56, 0x9a, 0x7a, 0xfd, 0xe1, 0x0f, 0xc8, 0x5c, 
[1648121158.324890][13467:13467] CHIP:IN: 0xa8, 
[1648121158.324903][13467:13467] CHIP:IN: Generated compressed fabric ID
[1648121158.324908][13467:13467] CHIP:IN: 0x0b, 0x65, 0x94, 0x24, 0xb8, 0xc7, 0x1a, 0x38, 
[1648121158.324999][13467:13467] CHIP:IN: Loading from storage for fabric index 3
[1648121158.325093][13467:13467] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1648121158.325100][13467:13467] CHIP:IN: 0x04, 0xe5, 0xbe, 0x64, 0xbe, 0x2b, 0x1c, 0x97, 
[1648121158.325104][13467:13467] CHIP:IN: 0x76, 0x89, 0xfc, 0xfe, 0xd8, 0x7b, 0x56, 0x4b, 
[1648121158.325107][13467:13467] CHIP:IN: 0x05, 0x9a, 0x89, 0x18, 0xf3, 0x32, 0x4b, 0x70, 
[1648121158.325111][13467:13467] CHIP:IN: 0xee, 0x90, 0x95, 0x4b, 0x89, 0xb0, 0x0f, 0x0e, 
[1648121158.325115][13467:13467] CHIP:IN: 0x9f, 0x03, 0x2a, 0x5f, 0xf5, 0x25, 0xdd, 0xcc, 
[1648121158.325118][13467:13467] CHIP:IN: 0xae, 0xaf, 0x5e, 0x03, 0x26, 0xfd, 0x31, 0x2f, 
[1648121158.325122][13467:13467] CHIP:IN: 0xa4, 0x0a, 0xfc, 0xda, 0x30, 0x2a, 0x0e, 0xc5, 
[1648121158.325125][13467:13467] CHIP:IN: 0x56, 0x9a, 0x7a, 0xfd, 0xe1, 0x0f, 0xc8, 0x5c, 
[1648121158.325128][13467:13467] CHIP:IN: 0xa8, 
[1648121158.325139][13467:13467] CHIP:IN: Generated compressed fabric ID
[1648121158.325144][13467:13467] CHIP:IN: 0x79, 0xc5, 0xf0, 0x8e, 0x6a, 0xd3, 0x1d, 0xef, 
[1648121158.325233][13467:13467] CHIP:IN: Loading from storage for fabric index 4
[1648121158.325251][13467:13467] CHIP:IN: Loading from storage for fabric index 5
[1648121158.325267][13467:13467] CHIP:IN: Loading from storage for fabric index 6
[1648121158.325282][13467:13467] CHIP:IN: Loading from storage for fabric index 7
[... 218 lines cut ...]
[1648121158.563562][13467:13472] CHIP:EM: Flushed pending ack for MessageCounter:10102540 on exchange 57836i
[1648121158.563740][13467:13467] CHIP:CTL: Shutting down the commissioner
[1648121158.563814][13467:13467] CHIP:CTL: Shutting down the controller
[1648121158.563841][13467:13467] CHIP:CTL: Shutting down the commissioner
[1648121158.563913][13467:13467] CHIP:CTL: Shutting down the controller
[1648121158.563948][13467:13467] CHIP:IN: Expiring all connections for fabric 1!!
[1648121158.563997][13467:13467] CHIP:CTL: Shutting down the commissioner
[1648121158.564037][13467:13467] CHIP:CTL: Shutting down the controller
[1648121158.564053][13467:13467] CHIP:IN: Expiring all connections for fabric 2!!
[1648121158.564088][13467:13467] CHIP:CTL: Shutting down the commissioner
[1648121158.564123][13467:13467] CHIP:CTL: Shutting down the controller
[1648121158.564139][13467:13467] CHIP:IN: Expiring all connections for fabric 3!!
[1648121158.564169][13467:13467] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1648121158.565262][13467:13467] CHIP:DMG: IM WH moving to [Uninitialized]
[1648121158.565303][13467:13467] CHIP:DMG: IM WH moving to [Uninitialized]
[1648121158.565319][13467:13467] CHIP:DMG: IM WH moving to [Uninitialized]
[1648121158.565332][13467:13467] CHIP:DMG: IM WH moving to [Uninitialized]
[1648121158.565391][13467:13467] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1648121158.565659][13467:13467] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-QMYN8u)
[1648121158.566325][13467:13467] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648121158.566389][13467:13467] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1648121158.566414][13467:13467] CHIP:DL: Inet Layer shutdown
[1648121158.566433][13467:13467] CHIP:DL: BLE shutdown
[1648121158.566449][13467:13467] CHIP:DL: System Layer shutdown
[1648121158.566530][13467:13467] CHIP:TOO: Run command failure: IM Error 0x00000587: General error: 0x87

yufengwangca, 2022-04-07:
This is because this attribute is not writable

 <attribute side="server" code="0x02" define="SUPPORTED_LOCALES" type="ARRAY" entryType="CHAR_STRING" length="254" writable="false" optional="false">SupportedLocales</attribute>

cjandhyala, 2022-04-08:
@yufengwangca The error code we were supposed to get is "UNSUPPORTED_WRITE(0x88)" , instead we are getting "CONSTRAINT_ERROR(0x87)"

andy31415, 2022-05-13:
Seems like the same as #17598

vivien-apple, 2022-05-18:
It happens because the code is going thought some old ember code path at Failed to prepare data to write: ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:866: CHIP Error 0x0000002F: Invalid argument

For some reason I can not reproduce while trying with #18561 (which is itself on top of https://github.com/project-chip/connectedhomeip/issues/17633)
I will wait for those to land and see if that still happens. There is no reason that this attributes goes into our compatibility layer..

vivien-apple, 2022-05-18:
Ok... Now I know why it was working with https://github.com/project-chip/connectedhomeip/pull/18561.
The command used in the description is assuming that "SupportedLocales" have the correct attribute id per spec. It does not currently as mentioned in #17633 and if the command is updated to represent that so it becomes `chip-tool any write-by-id 0x002B 2 newsupportedlocales 3 0` then `UNSUPPORTED_WRITE` is received.

Closing since the real issue is #17633.

closed 2022-05-18 by vivien-apple