project-chip/connectedhomeip #16627: Attribute "label list" of User Labelcluster showing success response instead of failure

manjunathambiger23, 2022-03-24:
Summary:
               As per test-plan "label list" needs to write the value expected outcome is CONSTRAINT_ERROR but showing success response instead of failure response

Test-plan screenshot:
 [image]

Spec screenshot:
[image]

Execution logs and screenshot:
[image]
[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/8342967/exe_log.txt)

manjunath-grl, 2022-05-10:
This issue was created by me using a account that i had to delete due to access issues, I will continue to follow these issues with my current id.

woody-apple, 2022-05-26:
Spec Review: @cjandhyala can you please follow up to confirm that this still happens, and/or how to update the tests.

cjandhyala, 2022-05-27:
@manjunath-grl , is this still an issue ?

manjunath-grl, 2022-05-30:
@cjandhyala  testplan for Label cluster has been changed and removed this particular steps can you please close this issue

cjandhyala, 2022-05-30:
closing per @manjunath-grl request