project-chip/connectedhomeip #16633: Nordic board with lighting example app doesn't power up - ZEPHYR FATAL ERROR 0

kean-apple, 2022-03-24:
Nordic board doesn't power up with recent Matter SHA

SHA: 011efcf02bbd1e9ca3fd2c50f9084d3755ba13e5

1. Compile Nordic board eg. git pull newer open source, confirm build environment  & run python3 scripts/setup/nrfconnect/update_ncs.py --update
2. Go to  /connectedhomeip/examples/lighting-app/nrfconnect - build and flash Nordic board 
west build -b nrf52840dk_nrf52840 --pristine always
west flash --erase
 
On power up, seeing the following error:
uart:~$ I: 8 Sectors of 4096 bytes
I: alloc wra: 0, fe8
I: data wra: 0, 0
*** Booting Zephyr OS build v2.7.99-ncs1-1  ***
D: 125 [DL]Boot reason: 1
I: 230 [DL]BLE address: ED:D6:F9:BD:7A:0C
I: SoftDevice Controller build revision: 
I: 0e e7 c5 66 67 18 3c ac |...fg.<.
I: b3 d2 cc 81 a3 dc f1 c0 |........
I: c0 36 02 22             |.6."    
I: 254 [DL]OpenThread started: OK
I: 257 [DL]Setting OpenThread device type to FULL END DEVICE
I: Swap type: none
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x00000001
E: r3/a4:  0x00000000 r12/ip:  0x0003b19d r14/lr:  0x00097c21
E:  xpsr:  0x80000000
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200070f0 (main)
E: Halting system

Did not see this on SHA from 3/22:  32f23fa1cf1d1e42a3eedc99822fede7134e66a2

bzbarsky-apple, 2022-03-24:
@Damian-Nordic

Damian-Nordic, 2022-03-24:
Fix is coming: https://github.com/project-chip/connectedhomeip/pull/16609

closed 2022-03-25 by bzbarsky-apple