project-chip/connectedhomeip #16641: Servers over-zealously restricts simultaneous writes to the same fabric-scoped list

mrjerryjohns, 2022-03-24:
#### Problem

Today, the IM prevents two clients from writing to the same attribute if one attribute write is still in progress (i.e chunked).

For fabric-scoped lists, writes are by design isolated by fabric, so doing so is over-zealous and in the worst case, can introduce denial-of-service issues.

 #### Proposed Solution

Relax the requirement for fabric-scoped lists.

mrjerryjohns, 2022-04-11:
Assigning this to you @erjiaqing since you touched this area last.

erjiaqing, 2022-04-13:
I guess we need to update ZAP to include "fabric scoped" bits to `EmberAfAttributeMetadata` since this infomation lives deeply in the cluster code.

The actual change could be quite simple, add a "IsFabricScoped" flag to `HasConflictWriteRequests` and an extra check in that function.

Some issues are:
1. Does this apply to fabric scoped lists?
2. How does the DoS affect the device? By the long timeout for interaction model messages?

mrjerryjohns, 2022-04-18:
1. Yes, only fabric-scoped lists.
2. One fabric can keep writing to a fabric-scoped list, preventing another fabric from writing to that list (with current code) till the first one is complete.

closed 2022-12-21 by stale[bot]