project-chip/connectedhomeip #16653: min report interval 0 subscription loses CurrentLevel reports

mspang, 2022-03-25:
#### Problem

Setting the minimal report interval to zero so that updates arrive in the minimal possible time causes level control cluster reports to be lost.

```
In [1]: node = 0x3e8

In [2]: sub = await devCtrl.ReadAttribute(node, [Clusters.OnOff.Attributes.OnOff, Clusters.LevelControl.Attributes.CurrentLevel], reportInterval=(0, 120))
2022-03-24 22:21:15 yuif chip.SC[372088] ERROR The device does not support GetClock_RealTimeMS() API. This will eventually result in CASE session setup failures.

In [3]: await devCtrl.SendCommand(node, 1, Clusters.OnOff.Commands.On())

Attribute Changed:
{
│   'Endpoint': 1,
│   'Attribute': <class 'chip.clusters.Objects.OnOff.Attributes.OnOff'>,
│   'Value': True
}
Attribute Changed:
{
│   'Endpoint': 1,
│   'Attribute': <class 'chip.clusters.Objects.LevelControl.Attributes.CurrentLevel'>,
│   'Value': 1
}
In [4]: sub = await devCtrl.ReadAttribute(node, [Clusters.OnOff.Attributes.OnOff, Clusters.LevelControl.Attributes.CurrentLevel])
Attribute Changed:
{
│   'Endpoint': 1,
│   'Attribute': <class 'chip.clusters.Objects.LevelControl.Attributes.CurrentLevel'>,
│   'Value': 254
}

In [5]: await devCtrl.SendCommand(node, 1, Clusters.OnOff.Commands.Off())

Attribute Changed:
{
│   'Endpoint': 1,
│   'Attribute': <class 'chip.clusters.Objects.OnOff.Attributes.OnOff'>,
│   'Value': False
}
In [6]: sub = await devCtrl.ReadAttribute(node, [Clusters.OnOff.Attributes.OnOff, Clusters.LevelControl.Attributes.CurrentLevel])
Attribute Changed:
{
│   'Endpoint': 1,
│   'Attribute': <class 'chip.clusters.Objects.LevelControl.Attributes.CurrentLevel'>,
│   'Value': 254
}
```

@mrjerryjohns @erjiaqing

mrjerryjohns, 2022-03-25:
Yeah, seems something is wonky in the dirty management and report generation logic...

mrjerryjohns, 2022-03-25:
We weren't running the reporting engine upon receiving a status response to the last report in the exchange. This would happen automatically with a min interval != 0, since that schedules a timer after the report gets sent (assuming the timer fires after we receive a status response, which it should in most cases...)

Will put out a fix tomorrow with tests...

PR #17426 (2022-05-04): Fix bug with ReportingEngine not being run with short/0 min-reporting intervals
#### Problem

Fixes #16653. 

We weren't seeing some attribute changes make their way to a client through a report on a subscription. This only happens when two things happen at the same time:

1. When two attributes change in quick succession, with the second attribute changing in subsequent execution context than the first (as is the case when you issue the Off() command to the OnOff cluster, where the state of the on-off cluster changes to off, but the level cluster only changes to 254 in a subsequent execution  [... cut]
closed 2022-05-04 by mrjerryjohns