project-chip/connectedhomeip #16664: Pigweed code breaks with python 3.10

jimlyall-q, 2022-03-25:
#### Problem

 * Expected - Python 3 update would be handled seamlessly
 * Actual - See screenshot
 * Steps - upgrade to Jellyfish try to activate master
 * Config - N/A
 * screenshots, images, logs, etc.
 * [image]

 >

 #### Proposed Solution
Change `third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py`
```
    # We expect Python 3.8, but if it came from CIPD let it pass anyway.
    if ('3.8' not in version and '3.9' not in version
            and 'chromium' not in version):
        print('=' * 60, file=sys.stderr)
        print('Unexpected Python version:', version, file=sys.stderr)
        print('=' * 60, file=sys.stderr)
        return False
```
There may be a better way to do this ^^
It doesn't appear to choose the recognised default version of python for the shell as set by update-alternatives

vringar, 2022-04-26:
[This](https://bugs.chromium.org/p/pigweed/issues/detail?id=616&q=Python&can=2) seems to be the relevant upstream issue.
Let's see how long they take with this.

brosahay, 2022-05-02:
I faced the same problem when I moved to Ubuntu 22.04 LTS. For the time being I got it to work using these commands
[install-python-3.9.12.sh](https://gist.github.com/brosahay/1cf580bf76b97686e6f951b1bb0275e6)

ashegheparyshun, 2022-05-03:
@brosahay 
It didn't worked for me. Do you have any idea?
[image]

ashegheparyshun, 2022-05-03:
@brosahay 
Hi, Thanks for your respond.
[image]
Am I doing something wrong?

brosahay, 2022-05-03:
> @brosahay Hi, Thanks for your respond. [image] Am I doing something wrong?

Hey, you can try these commands and sharing the output
```shell
git reset --hard
git clean -dfx
git submodule update
git status
```
This will give you the idea if the repo is clean and reset to the HEAD, since `python --version' is working, my best guess would be that the enviroment is stuck at python 3.10.4.

If everything fails recloning the repo could work, but I know its a lot of data :(

chadnorvell, 2022-05-03:
Hey everyone, I'm on the Pigweed team and I'm looking into this.

You shouldn't have any issues bootstrapping the project with a system Python version 3.10. I've tested that myself successfully.

This section of code in `third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py`:

```
    # We expect Python 3.8, but if it came from CIPD let it pass anyway.
    if ('3.8' not in version and '3.9' not in version
            and 'chromium' not in version):
        print('=' * 60, file=sys.stderr)
        print('Unexpected Python version:', version, file=sys.stderr)
        print('=' * 60, file=sys.stderr)
        return False
```

... isn't constraining the versions of your *system* Python, but rather the version of Python that is vendored in the Pigweed environment. In upstream Pigweed, that's defined [here](https://github.com/google/pigweed/blob/2e14574efe7d022966fc971151f0dfbbd81c85dc/pw_env_setup/py/pw_env_setup/cipd_setup/python.json). We expect to pull down Python 3.9 for the Pigweed environment, or 3.8 for ARM64 Macs. Matter overrides that for some platforms [here](https://github.com/project-chip/connectedhomeip/blob/master/scripts/python.json), but that still points to 3.9.

So while I also want to fully support the use of Python 3.10 in the Pigweed environment, at the moment I'm curious about why your bootstrapping script is either using your system Python 3.10 or pulling Python 3.10 from CIPD. To that end, could you apply this patch to `third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py`:

```
  diff --git a/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py b/pw_env_setup/py/pw_env_setup/virtualenv_setup/instal  l.py
  index b8841b99..700db500 100644
  --- a/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py
  +++ b/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py
  @@ -147,6 +147,10 @@ def install(  # pylint: disable=too-many-arguments

       version = subprocess.check_output(
           (python, '--version'), stderr=subprocess.STDOUT).strip().decode()
  +    which = subprocess.check_output(
  +        ('which', python), stderr=subprocess.STDOUT).strip().decode()
  +    print('\n' + which)
  +    print(version)
       # We expect Python 3.8, but if it came from CIPD let it pass anyway.
       if ('3.8' not in version and '3.9' not in version
               and 'chromium' not in version):
```

... and try bootstrapping again after removing your `environment` directory, and report back with the output?

vringar, 2022-05-03:
@chadnorvell will do that tomorrow. Thanks for looking into this.

vringar, 2022-05-04:
```
Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (1m55.8s)
  Setting up Python environment.....[|]
/usr/bin/python3
Python 3.10.4
============================================================
Unexpected Python version: Python 3.10.4
============================================================
failed (0.1s)
```

ashegheparyshun, 2022-05-04:
@chadnorvell
I deleted connectedhomeip folder and tried everything from the beginning the result is just like before.
 
[image]

chadnorvell, 2022-05-04:
The issue appears to be that bootstrap is using your system Python instead of the one vendored in the Pigweed environment. After the bootstrap is finished, do you have a `pigweed-venv` directory in your `.environment` or `environment` directory? If so, does the `bin` inside of it look like it has Python binaries? Also, what OS and CPU architecture are you using?

ashegheparyshun, 2022-05-05:
@chadnorvell 
[image]
[image]
[image]
As you can see the link here are pointing to the links located in usr/local/bin

ashegheparyshun, 2022-05-05:
In conclusion, before bootstrap follow this steps:
1. Run this code [install-python-3.9.12.sh](https://gist.github.com/brosahay/1cf580bf76b97686e6f951b1bb0275e6).
2. Add this patch https://github.com/project-chip/connectedhomeip/issues/16664#issuecomment-1116540876
3. Run `source scripts/activate.sh` from `connectedhomeip`.
4. After failure go to the directory mentioned and remove all links.
[image]
For example in this case, remove all links located in ` /usr/local/bin` directory.
5. Run these commands.
`sudo ln -s /opt/python/3.9.12/bin/python3.9 /usr/local/bin/python3`
`sudo ln -s /opt/python/3.9.12/bin/python3.9 /usr/local/bin/python`
`sudo ln -s /opt/python/3.9.12/bin/pip3 /usr/local/bin/pip3`
`sudo ln -s /opt/python/3.9.12/bin/pip /usr/local/bin/pip`
`sudo ln -s /opt/python/3.9.12/bin/pydoc3.9 /usr/local/bin/pydoc`
`sudo ln -s /opt/python/3.9.12/bin/idle /usr/local/bin/idle`
`sudo ln -s /opt/python/3.9.12/bin/python-config /usr/local/bin/python-config`
6. Bootstrap again.
Thanks to @brosahay and @chadnorvell .

tcarmelveilleux, 2022-05-05:
> In conclusion, before bootstrap follow this steps:
> 
> 1. Run this code [install-python-3.9.12.sh](https://gist.github.com/brosahay/1cf580bf76b97686e6f951b1bb0275e6).
> 2. Add this patch [Pigweed code breaks with python 3.10 #16664 (comment)](https://github.com/project-chip/connectedhomeip/issues/16664#issuecomment-1116540876)
> 3. Run `source scripts/activate.sh` from `connectedhomeip`.
> 4. After failure go to the directory mentioned and remove all links.

This is possibly unsafe, and not a real fix, since it requires running a binary to override system defaults. System defaults may be IT policy of a given organization

vringar, 2022-05-05:
Also this may interfere with your package managers assumptions about your environment.

rosahay-silabs, 2022-05-10:
@chadnorvell were you able to find out the reason pigweed bootstrap is not working as expected ?

locomuco, 2022-05-20:
Using the docker container for development mitigates this problem ([VS Code Instructions](https://github.com/project-chip/connectedhomeip/blob/master/docs/VSCODE_DEVELOPMENT.md))

jonsmirl, 2022-05-26:
```
$ sudo add-apt-repository ppa:deadsnakes/ppa
$ sudo apt update && sudo apt upgrade
$ sudo apt install python3.9 python3.9-dev python3.9-venv
```

This should have worked but it does't.
`$ export PW_BOOTSTRAP_PYTHON=python3.9`

Get annoyed at pigweed and simply hard code it:
```
jonsmirl@ares:~/aosp/chip/third_party/pigweed/repo$ git diff
diff --git a/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py b/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py
index b8841b99..8bfd447d 100644
--- a/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py
+++ b/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py
@@ -145,6 +145,7 @@ def install(  # pylint: disable=too-many-arguments
 ):
     """Creates a venv and installs all packages in this Git repo."""
 
+    python='python3.9'
     version = subprocess.check_output(
         (python, '--version'), stderr=subprocess.STDOUT).strip().decode()
     # We expect Python 3.8, but if it came from CIPD let it pass anyway.
```

Then bootstrap will work:
`source ./scripts/bootstrap.sh`

presto8, 2022-06-06:
Tested @jonsmirl's solution on Ubuntu 22.04 (Raspberry Pi) and it works well. Without installing python3.9 and editing install.py, active.sh still fails with the python3.10 message.

cduvenhorst, 2022-06-22:
> @tcarmelveilleux @vringar 
> 
> Still waiting for the real solution.

Solution ahead: If Tornadoweb is [updated](https://github.com/tornadoweb/tornado/issues/3057#issuecomment-1161021068)for Python 3.10.x, there should be no reason not supporting it in pigweed. 

However, matters requirements for numpy must be higher than version 1.22.

Ursescu, 2022-06-23:
Hi, I had the same problem on my Ubuntu 22.04 box. I found a neat solution to download and configure Python 3.8 using [pyenv] (https://github.com/pyenv/pyenv), and it works perfectly. I'm writing this in the hopes that someone will find it useful.

jonsmirl, 2022-06-23:
If you change the global python version, sure CHIP will build and work, but there are a lot of packages in Ubuntu 22 which assume Python 3.10. By reverting the system Python back to 3.8 you are going to break those other packages.  Maybe you don't use those other packages, but if you do, you are going to end up with something that is hard to debug when they fail to work.

The deadsnakes technique installs 3.9 along side 3.10 so that both are on your system. Then CHIP is adjusted to use the python3.9 install with that small edit posted above.

Ursescu, 2022-06-23:
@jonsmirl I use pyenv to override the Python version only in the current shell, but I agree, even in this case, it can lead to bad things.

presto8, 2022-06-23:
In one sense, the use of pyenv (or any virtualenv) is the correct approach for locking a dependency like this: if a program specifically is asking for python3.9, then just replacing it with 3.10 is not necessarily the right thing to do. In this case it seems to work fine though.

But this project already uses virtualenv I believe, so I think the best approach here would be to configure that original existing virtualenv to install the correct version of Python that it wants to use?

The issue I had was the guide I was following (in the official documentation I think) said to use a specific version of Ubuntu. However, that version was not an LTS release and is no longer available. So I went to the next LTS release after the one that was recommended, which was 22.04, which ships with Python 3.10 now, and hence the problem :)

cduvenhorst, 2022-06-25:
> ... isn't constraining the versions of your _system_ Python, but rather the version of Python that is vendored in the
> Pigweed environment. In upstream Pigweed, that's defined [...].
> We expect to pull down Python 3.9 for the Pigweed environment, or 3.8 for ARM64 Macs. Matter overrides that for some
> platforms [here](https://github.com/project-chip/connectedhomeip/blob/master/scripts/python.json), but that still points to
>3.9.

"Overrides" - yes. BUT: Linux is missing as an overridden platform in matter and since there is no Inheritance, the system python version is used (no python version restriction for the venv anymore).

cduvenhorst, 2022-06-25:
Ok. Here is a solution for all the folks who do not want to use those fine docker images, which works at least for me (Ubuntu 22.04 LTS):

1) Unzip file to an empty directory
[Matter_python3_10_x.zip](https://github.com/project-chip/connectedhomeip/files/8984410/Matter_python3_10_x.zip)

2) Download and patch matter
`./patch_for_py_3.10.x.sh`

3) Change to the matter directory and activate it.
`cd matter && . ./scripts/activate.sh`

presto8, 2022-06-25:
@cduvenhorst , using docker sounds good! I didn't even know that was really an option. Should the building guide be updated to say this is the preferred method?  https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

turon, 2022-06-29:
For my environment, I was able to restore it by reverting this [pigweed commit](https://pigweed-review.googlesource.com/c/pigweed/pigweed/+/98520/12/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py#149) and adding 3.10 to the [acceptlist](https://pigweed-review.googlesource.com/c/pigweed/pigweed/+/98520/12/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py#170).

nekleo, 2022-06-30:
@chadnorvell 
> The issue appears to be that bootstrap is using your system Python instead of the one vendored in the Pigweed environment. After the bootstrap is finished, do you have a `pigweed-venv` directory in [...]

Yes, bootstrap seems to be using system Python. #20158 will allow 3.10 to work, but seems that it'd be best if the bootstrap could use the python installed by pigweed. After bootstrap is finished, I have a `pigweed-venv` in `.environment`.

CPU=x86_64
OS=Debian GNU/Linux rodete

```
% ls -la pyth*
lrwxr-xr-x  1 xxx  primarygroup   7 Jun 14 17:04 python -> python3
lrwxr-xr-x  1 xxx  primarygroup  86 Jun 14 17:04 python3 -> /Users/xxx/Projects/connectedhomeip/.environment/cipd/packages/python/bin/python3
lrwxr-xr-x  1 xxx  primarygroup   7 Jun 14 17:04 python3.9 -> python3
% 
```

mikaelhm, 2022-07-25:
Like @nekleo reported above, this is no longer an issue.

I just verified that I am able to complete `./scripts/bootstrap.sh` without issues on raspberry pi 4 with Ubuntu 22.04, which comes with Python 3.10.

mikaelhm, 2022-07-29:
This can be closed, verified no longer an issue.

franck-apple, 2022-07-29:
Closing per above comment.

closed 2022-07-29 by franck-apple