project-chip/connectedhomeip #16678: chip-device-ctrl > set-pairing-wifi-credential <ssid> <psk> --> Error - 'bytes' object has no attribute 'encode'

HeneryH, 2022-03-25:
#### Problem
This seems new to me.  Haven't had this problem before.  Not sure if anything changed or anyone else is seeing this.

```
chip-device-ctrl > set-pairing-wifi-credential My-SSID 10151930
'bytes' object has no attribute 'encode'

chip-device-ctrl > set-pairing-wifi-credential My-SSID '10151930'
'bytes' object has no attribute 'encode'

chip-device-ctrl > set-pairing-wifi-credential My-SSID "10151930"
'bytes' object has no attribute 'encode'

chip-device-ctrl > set-pairing-wifi-credential "My-SSID" "qwe10151930"
'bytes' object has no attribute 'encode'

``` 
#### Proposed Solution
<suggested fix, suggested enhancement>

HeneryH, 2022-03-25:
Seems like a string is being mis-typed as a byte[] or vise versus.

timredfern, 2022-05-11:
I see this problem too with chip-device-ctrl, I'm using commit 44dd01b9fa49b2a31a7d8ba52084cef7d9e73e17 from May 9, OSX 12.3.1

caipiblack, 2022-05-15:
Same here on [TE9] tag.

hidaris, 2022-06-21:
I fix it in https://github.com/project-chip/connectedhomeip/pull/19792 You can try it to see if it solves your problem

PR #19792 (2022-06-21): remove duplicate str encode in py SetWiFiCredentials
#### Problem
What is being fixed?  Examples:
* Fixes #16678 , str encode duplicate in https://github.com/project-chip/connectedhomeip/blob/master/src/controller/python/chip-device-ctrl.py#L951 and https://github.com/project-chip/connectedhomeip/blob/master/src/controller/python/chip/ChipDeviceCtrl.py#L348
#### Change overview
What's in this PR
remove duplicate str encode

#### Testing
How was this tested? (at least one bullet point required)
* If manually tested, what platforms controller and device platforms were manually tested, and how?
A [... cut]
closed 2022-06-21 by andy31415