project-chip/connectedhomeip #16691: Support all attributes by Darwin CHIPDevice subscribeWithQueue API

kpark-apple, 2022-03-26:
#### Problem
Current CHIPDevice subscribeWitthQueue: API only supports known standard attributes. For manufacturer specific extensions or unimplemented standard attributes, subscribeAttributeWithEndpointId:... API can be used. However, in such a case, it forces both of them to set keepPreviousSubscriptions option in order not to cancel each other, limiting the flexibility of application to clear up previous subscriptions.

 #### Proposed Solution
Add a generic report handler argument of CHIPDeviceResponseHandler type to subscribeWithQueue API in addition to the existing standard attribute report handler. Note that if standard event handler argument is added to the API, the same generic report handler can be used to receive any event as well as any attribute.

bzbarsky-apple, 2022-03-28:
@jtung-apple Note the bit about events above.

woody-apple, 2022-06-14:
Darwin Review: Note to self, update this to cover current scope for subscriptions.

jtung-apple, 2022-08-03:
Renamed issue to better reflect what this is tracking.

Details: https://github.com/project-chip/connectedhomeip/pull/20519 implemented generic attributes, but is still missing generic events, as well as other TODOs for the synchronous-read API. Updating this issue to explicitly track these items.

jtung-apple, 2022-08-16:
Closing because #21566 and #21757 are in now.

PR #17436 (2022-04-25): Darwin framework: add replacement subscribe method that includes event report
#### Problem
* Darwin framework CHIPDevice was missing event subscription. This PR aims to add it
Fixes #16691

#### Change overview
* zapt templates to generate Obj-C objects corresponding to C++ event structs
* zapt template to populate these objects from events TLVReader
* new CHIPDevice subscribeWithQueue method with `eventReporterHandler` and logic to hook up the handler with the callback mechanism

#### Testing
Work-in-progress - will update testing details later.
closed 2022-08-16 by jtung-apple