project-chip/connectedhomeip #16694: After a4010a commit, Android Controller always fails to pair

RudyGOGOGO, 2022-03-26:
With a4010a commit, the Android Controller can pair with my ESP32-DevKitC, however, after I update the local repo with all new changes after a4010a commit, my Android Controller always fails to pair with the device
Basic info about my experiment
> 1)Computer System: Ubuntu 20.04
> 2)Device: ESP32-DevKitC-VE
> 3)Rendezvous Mode: BLE
> 4)The latest version of Android Controller
[FailedToPairLog.txt](https://github.com/project-chip/connectedhomeip/files/8354701/FailedToPairLog.txt)

RudyGOGOGO, 2022-03-26:
@bzbarsky-apple @vijs I really appreciate your quick response about https://github.com/project-chip/connectedhomeip/issues/16600, and with latest changes, the issue about PAA is gone, the CHIP Tool Command Line without `--paa-trust-store-path` works well. However, I am facing this Android Controller Pairing issue which does not exist before those new changes are made.
I am using the same Android Controller.

bzbarsky-apple, 2022-03-26:
That log shows the node being commissioned until it's advertising on the operational network...

Past that, it's hard to say what's not working; "fails to pair" is not very specific.

If you have a changeset that does work for you and one that does not it might be useful to do a git bisect to find out which exact commit caused the problem.

Past that, maybe @austinh0 might know something about the Android controller and recent changes to it...

RudyGOGOGO, 2022-03-26:
Thanks for your quick response, I feel sooooo sorry to keep bothering you~
Tomorrow, I will do more tests on Android Controller. Later on, I will rephrase the title in order to make it specific.