project-chip/connectedhomeip #16705: Add UNSUPPORTED_NODE support

vivien-apple, 2022-03-28:
#### Problem

`UNSUPORTED_NODE` has been added in the spec by https://github.com/CHIP-Specifications/connectedhomeip-spec/commit/f56e0bab1bd902428a1336b812c6b76a464c27ff

It needs to be supported into the SDK.

 #### Proposed Solution
 * Add supports to it

vivien-apple, 2022-04-13:
Assigning to you @mrjerryjohns - that sounds like something you know about.

mrjerryjohns, 2022-04-20:
A condition that results in this error won't happen in the SDK today since we're not implementing proxies for 1.0 (that's the only place this could happen).

yunhanw-google, 2022-04-25:
@vivien-apple  close it?

franck-apple, 2022-04-27:
> @vivien-apple close it?

looks more appropriate to push it to v1.x?

Rajashreekalmane, 2023-08-11:
Still this issue is not resolved, hence reopening this issue

Rajashreekalmane, 2024-06-28:
Test step has been removed from test-plan 
https://github.com/CHIP-Specifications/chip-test-plans/pull/4194/