project-chip/connectedhomeip #16716: unlockwithTimeout command failing when executed

manjunathambiger23, 2022-03-28:
Summary:
              unlockwithTimeout command showing failure response instead of success response when executed.

Spec screenhot:
[image]

Execution screenshot and logs: 
[image]
[unlockwithTimeout_command.txt](https://github.com/project-chip/connectedhomeip/files/8362806/unlockwithTimeout_command.txt)

bzbarsky-apple, 2022-03-28:
0x81 is UNSUPPORTED_COMMAND.

Note that "Unlock with Timeout" is an optional door lock command; servers do not have to support it.

Which server application is being used?

In `examples/door-lock-app/door-lock-common/door-lock-app.zap` this command is not enabled.  In `examples/all-clusters-app/all-clusters-common/all-clusters-app.zap` this command is likewise not enabled.

manjunathambiger23, 2022-03-29:
@bzbarsky-apple we are using Door Lock app ,we try enabling that command and let you know

PR #17226 (2022-04-20): Door lock fixes for testing
#### Problem
* Fixes Door Lock issues for testing.
* Fixes #16606
* Fixes #16810 
* Fixes #16716 
* Fixes #16702 
* Fixes #16672 
* Fixes #16656 

#### Change overview
- Updated Zap files to include the missing clusters and attributes.
- Updated Unlock and Lock behavior to not accept unlocks or locks if pin is not correct. See issue #16702 

#### Testing
- Obtained the YAMLs having issues
- Generated code
- Compiled
- Ran the YAML tests agains the all-clusters-app.
closed 2022-04-20 by bzbarsky-apple