project-chip/connectedhomeip #16725: FeatureMap attribute should be present on all clusters

bzbarsky-apple, 2022-03-28:
#### Problem
Spec changed in https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5000 to require feature map on all clusters, with the value set to 0 if the cluster has no optional features.

 #### Proposed Solution
We have two options.

Option one is to do nothing in the SDK, update the XML to mark FeatureMap as a required global attribute, update all the ZAP configs to enable feature map on all clusters, and call it a day for 1.0.

Option two is to do something in the attribute read and wildcard expansion codepaths where we always add in the FeatureMap attribute if it's not already present (similar to what we do for the command list and attribute list, attributes, but a bit more complicated because those are _never_ present in the actual endpoint config, whereas feature map sometimes would be).

tehampson, 2022-05-09:
I will have a PR coming soon that does option 1

mrjerryjohns, 2022-05-19:
Presumably a minimal version of this will be solved by #18213?

woody-apple, 2022-05-19:
Cert Blocker Review: @tehampson once complete, can you please file a followup issue to ensure the XML is spec conformant and valid.

tehampson, 2022-05-19:
> Cert Blocker Review: @tehampson once complete, can you please file a followup issue to ensure the XML is spec conformant and valid.

If my understanding is correct, the only change to make XML spec conformant for this is already in the PR:
https://github.com/project-chip/connectedhomeip/pull/18213/files#diff-0c8fe8ba57ad140fcc2d890c91031601ea3912191591d86fda602c03b8bc46f7

andy31415, 2022-05-19:
I believe this is just surfacing a general issue of "we need xml to match spec". It showed up during reviewing feature maps, with the question of "how do we make sure xml lists all possible features", however the generic followup should be "ensure XML matches spec".

We may also want to split the followup somehow, since each cluster could be checked individually.

bzbarsky-apple, 2022-05-20:
> Presumably a minimal version of this will be solved by https://github.com/project-chip/connectedhomeip/pull/18213?

Yes, that's implementing "option one".

PR #18213 (2022-05-23): Working python script to add manditory global attributes
#### Problem
* Fixes #16725
* Many cluster instances are giving 0x86(unsupported attribute) for mandatory attributes (ClusterRevision and FeatureMap).

#### Change overview
* Create python script that updates zap file to include 

#### Testing
* Manual:
  * Diffed newly created zap file and confirmed that ClusterRevision and FeatureMap attributes were added for all clusters that were previously missing these mandatory attributes
closed 2022-05-23 by bzbarsky-apple