project-chip/connectedhomeip #16748: RemoveFabric command crashes server when removing current fabric

bzbarsky-apple, 2022-03-29:
#### Problem
After #16098 we crash if we RemoveFabric and pass it the fabric index of the fabric the command is running on.

This happens because of this code in `SessionManager::PrepareMessage`:
```
        if (session->GetSecureSessionType() == SecureSession::Type::kCASE)
        {
            FabricInfo * fabric = mFabricTable->FindFabricWithIndex(session->GetFabricIndex());
            VerifyOrDie(fabric != nullptr);
            CryptoContext::BuildNonce(nonce, packetHeader.GetSecurityFlags(), messageCounter, fabric->GetNodeId());
```
when sending the response message for the RemoveFabric command.

 #### Proposed Solution
We need to either fix this code to get the node id from somewhere else (e.g. store it in the session instead of getting it from the fabric table every time) or change how exactly RemoveFabric works in some way.  I'm not sure what way.

@kghost @tcarmelveilleux @turon

turon, 2022-03-29:
I'd lean toward storing the node id redundantly in the session context.  That way we can support this "lingering session" use cases elegantly and decouple them from configuration updates that are occurring over the session.

kghost, 2022-04-13:
Discussed with Boris, we will achieve `RemoveFabric` in following actions:

1. Call OnFabricDeletedFromStorage
1. Send RemoveFabric response
1. Remove session associated with the fabric
1. Remove exchanges associated with the fabric
1. Call exchange->OnExchangeReleased to all exchange delegates
1. Remove the fabric object on next schedule (or next cold boot)

mlepage-google, 2022-04-14:
I can verify I hit this too while testing something. :-)

Damian-Nordic, 2022-04-21:
@kghost I agree with your solution, but I think you will need to make sure that the LEAVE event is generated prior to removing ACLs for the fabric. Otherwise, generating the report containing the LEAVE event will fail the permission check, right?

bzbarsky-apple, 2022-05-03:
Fixed in #17815.

closed 2022-05-03 by bzbarsky-apple