project-chip/connectedhomeip #16780: Inet::IPAddress::fromString should support parsing the scope ID as well

g-coppock, 2022-03-29:
#### Problem

Currently, providing a scoped IPv6 address (e.g. `fe80::3e61:5ff:fe0c:89f8%47`) where the network interface index is encoded in the IPv6 address, fails to parse. 

https://github.com/project-chip/connectedhomeip/blob/master/src/inet/IPAddress-StringFuncts.cpp#L85

 #### Proposed Solution

This method should be updated to parse and retain the scope ID, if present.

Maybe relevant discussion here? http://savannah.nongnu.org/task/?13106

wgtdkp, 2022-03-30:
@g-coppock It's probably not practical to add scope support to IPAddress parser directly because the IPAddress struct doesn't include the scope ID definition - It does make sense as scope is not part of IPv6 address but associated with the context where it is being used:

> The zone to which a particular non-global address pertains is not
   encoded in the address itself but determined by context, such as the
   interface from which it is sent or received.  Thus, addresses of a
   given (non-global) scope may be re-used in different zones of that
   scope.  For example, two different physical links may each contain a
   node with the link-local address fe80::1.

https://www.rfc-editor.org/rfc/rfc4007#section-5

So both the Lwip `ip6_addr_t` and linux `in6_addr` doesn't include scope but linux  `sockaddr_in6 ` does (https://man7.org/linux/man-pages/man3/inet_ntop.3.html).

While the CHIP [PeerAddress](https://github.com/project-chip/connectedhomeip/blob/0a760e4a4ce2ed7be6102e83b04554e9038f9c6f/src/transport/raw/PeerAddress.h) is something similar to `sockaddr_in6`, it should be easier to just add a string parser for `PeerAddress` and use it against `IPAddress::fromString` where a `PeerAddress` is required. Thoughts?

(Note: it seems mostly an `IPAddress` is used together with an additional `InterfaceId` parameter, for example https://github.com/project-chip/connectedhomeip/blob/b9d8ef17c54467d67d0bd86847f6dbbfd3d98cb3/src/inet/UDPEndPoint.cpp#L133

g-coppock, 2022-03-30:
That makes sense -- more specifically yeah I think we just need a way to go from a String to a scoped address. If `PeerAddress` makes more sense there I think that's fine.

closed 2022-10-08 by stale[bot]