project-chip/connectedhomeip #16818: Multi-admin should be done with unique fabric ID according to spec but in Raspberry Pi and MAC cases its done with same fabric ID

sowmyassp, 2022-03-30:
Issue : Multi-admin should be done with unique fabric ID but in chip-tool and MAC its showing same fabric ID when we read fabrics.

Commit ID used : 58cd5fb0ab06e2233ef52c1e575e0d750d13a833

Steps to reproduce :

DUT side : sudo rm -rf /tmp/chip_*
                  sudo ./chip-all-clusters-app --wifi
            
TH side : ./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840 (on TH1)
                ./chip-tool pairing open-commissioning-window 1 1 100 1000 3840(on TH1)
                ./chip-tool pairing manualcode 2 35484132896(on TH2)
                ./chip-tool operationalcredentials read fabrics 1 0 --fabric-filtered 0
                 ./chip-tool pairing open-commissioning-window 1 1 100 1000 3840(on TH1)
     
Spec reference : 
[image]

Issue screenshot : 
[image]

Logs : 
[1st_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8377829/1st_cntrl.txt)
[2nd_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8377832/2nd_cntrl.txt)

Rajashreekalmane, 2022-03-30:
Same issue facing in [TC-IDM-7.1] test-case 
In IDM-7.1 This test case will verify the report data messages for subscriptions requested from devices on different fabrics are handled correctly by the DUT. but in chip-tool and MAC its showing same fabric ID when we read
fabrics.

bzbarsky-apple, 2022-04-05:
> Multi-admin should be done with unique fabric ID

Within the scope of a given root public key.

The screenshow above shows both fabrics have fabric id 1 but different root public keys, so this is all fine.

closed 2022-04-05 by bzbarsky-apple