project-chip/connectedhomeip #16830: BLE Write method on Darwin platform conflicts with the specification

bluebin14, 2022-03-30:
#### Problem
Table 30. BTP GATT service specifies Write Without Response as the property for the C1 characteristic. Darwin platform code does the opposite, it uses Write With Response, which is better supported and provides transaction callback. There is a conflict between the specification and the implementation. 

 #### Proposed Solution
Change either the code or the spec or both, possibly allowing both Write With and Without Response. Note that changing the code would require extensive testing on all supported Darwin versions.

bauerschwan, 2022-04-01:
Just an additional thing we figured out: the BLE implementation on Raspberry Pi does exactly the same thing: it uses Write with response instead of Write without response. This is unfortunate because the Raspberry Pi does get used as a certification Test Harness and thus should comply to the spec.

turon, 2022-04-01:
It is true that the spec was updated to mandate use of "Write Without Response" with the assumption that it is more efficient.
It isn't clear whether any platforms were updated to use that, as the code historically used "Write With Response".
Supporting both would increase code size and complexity on all platforms.
If no platform implements "Write Without Response", the spec change path may be the lowest friction.

bauerschwan, 2022-04-04:
I highly doubt that implementing Write Without Response would lead into significant additional code size in case Write With Response was already implemented. 
Write Without Response is indeed more efficient and you see quite some speed improvement. But this is only used during BLE commissioning so in dayily use this speed improvement is not really visible.
The real issue is that the discrepancy between the spec and the implementation should get fixed before the next test event to make sure that this gets tested in the right way.

turon, 2022-04-13:
High level changes for Linux below.  Darwin changes need to be identified.

Replace string "request" with "command":
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/Helper.cpp#L1625

For notification change, replace "indicate" with "notify" and remove confirm:
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/Helper.cpp#L1238
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/Helper.cpp#L1267

After write is sent, make async call SendWriteRequestDone:
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/Helper.cpp#L1628

After notify is sent, make async call BluezCharacteristicConfirm:
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/Helper.cpp#L630

turon, 2022-04-13:
A partial review of platforms shows that WriteResp is used in most clients / platforms:

<google-sheets-html-origin><style type="text/css"></style>

BLE Role | Platform | C1 | C1 optionality | C2 | C2 optionality
-- | -- | -- | -- | -- | --
Central | Linux | WriteResp |   | Indication | none
Central | Darwin | WriteResp |   | Notify |  
Central | android | WriteResp |   | Notify |  
  |   |   |   |   |  
  |   |   |   |   |  
Peripheral | ESP32 | WriteResp |   | Notify | Indication?
Peripheral | Zephyr | Write w/o Con | WriteResp | Notify |

turon, 2022-05-02:
Spec change here: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5167

bluebin14, 2022-05-03:
Fixed and merged.

closed 2022-05-03 by bluebin14