project-chip/connectedhomeip #16834: start-udc-server command does not work on current master

cabin15, 2022-03-30:
#### Problem
Tested on master branch (58cd5fb0ab06e2233ef52c1e575e0d750d13a833)

chip-tool pairing start-udc-server

Expected result: Commissioner discovery should be available
Actual result: chip-tool crashes with seg fault.

Log
sungmokshin@sungmokshin-P425-KE1WK:~/master/connectedhomeip/out/host$ ./chip-tool pairing start-udc-server
[1648652954.010918][268478:268478] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-GAYhJg)
[1648652954.011180][268478:268478] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648652954.011196][268478:268478] CHIP:DL: NVS set: chip-counters/reboot-count = 6 (0x6)
[1648652954.011502][268478:268478] CHIP:DL: Got Ethernet interface: enp8s0
[1648652954.011706][268478:268478] CHIP:DL: Found the primary Ethernet interface:enp8s0
[1648652954.011915][268478:268478] CHIP:DL: Got WiFi interface: wlp7s0
[1648652954.011939][268478:268478] CHIP:DL: Failed to reset WiFi statistic counts
[1648652954.011981][268478:268478] CHIP:IN: UDP::Init bind&listen port=5542
[1648652954.012013][268478:268478] CHIP:IN: UDP::Init bound to port=5542
[1648652954.012022][268478:268478] CHIP:IN: UDP::Init bind&listen port=5542
[1648652954.012048][268478:268478] CHIP:IN: UDP::Init bound to port=5542
[1648652954.012058][268478:268478] CHIP:IN: TransportMgr initialized
[1648652954.012075][268478:268478] CHIP:DIS: Init fabric pairing table with server storage
[1648652954.041983][268478:268478] CHIP:IN: Loading certs from storage
[1648652954.042121][268478:268478] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1648652954.042158][268478:268478] CHIP:IN: 0x04, 0xff, 0xaa, 0xa7, 0xd2, 0x03, 0xe6, 0x11, 
[1648652954.042179][268478:268478] CHIP:IN: 0x30, 0x79, 0xf1, 0xa7, 0x87, 0xd6, 0xb9, 0x35, 
[1648652954.042199][268478:268478] CHIP:IN: 0xb0, 0x09, 0x58, 0x98, 0xba, 0x8a, 0xa8, 0xe5, 
[1648652954.042222][268478:268478] CHIP:IN: 0xd6, 0x92, 0x74, 0xfa, 0x15, 0x43, 0x1b, 0xeb, 
[1648652954.042248][268478:268478] CHIP:IN: 0x2e, 0xbe, 0x75, 0x7e, 0x84, 0x86, 0xf2, 0xbd, 
[1648652954.042270][268478:268478] CHIP:IN: 0x48, 0x20, 0xb3, 0xa6, 0xc8, 0xbd, 0x5d, 0xba, 
[1648652954.042289][268478:268478] CHIP:IN: 0x42, 0xfc, 0x20, 0xf1, 0x15, 0x2b, 0x91, 0xfc, 
[1648652954.042312][268478:268478] CHIP:IN: 0x0f, 0x2f, 0x68, 0x7c, 0x2d, 0x0a, 0x5a, 0x0c, 
[1648652954.042352][268478:268478] CHIP:IN: 0x83, 
[1648652954.042437][268478:268478] CHIP:IN: Generated compressed fabric ID
[1648652954.042470][268478:268478] CHIP:IN: 0xa1, 0x03, 0x84, 0x45, 0x63, 0x7f, 0xf8, 0x29, 
[1648652954.042776][268478:268478] CHIP:IN: Loading certs from storage
[1648652954.042858][268478:268478] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1648652954.042880][268478:268478] CHIP:IN: 0x04, 0xff, 0xaa, 0xa7, 0xd2, 0x03, 0xe6, 0x11, 
[1648652954.042896][268478:268478] CHIP:IN: 0x30, 0x79, 0xf1, 0xa7, 0x87, 0xd6, 0xb9, 0x35, 
[1648652954.042912][268478:268478] CHIP:IN: 0xb0, 0x09, 0x58, 0x98, 0xba, 0x8a, 0xa8, 0xe5, 
[1648652954.042927][268478:268478] CHIP:IN: 0xd6, 0x92, 0x74, 0xfa, 0x15, 0x43, 0x1b, 0xeb, 
[1648652954.042943][268478:268478] CHIP:IN: 0x2e, 0xbe, 0x75, 0x7e, 0x84, 0x86, 0xf2, 0xbd, 
[1648652954.042958][268478:268478] CHIP:IN: 0x48, 0x20, 0xb3, 0xa6, 0xc8, 0xbd, 0x5d, 0xba, 
[1648652954.042973][268478:268478] CHIP:IN: 0x42, 0xfc, 0x20, 0xf1, 0x15, 0x2b, 0x91, 0xfc, 
[1648652954.042989][268478:268478] CHIP:IN: 0x0f, 0x2f, 0x68, 0x7c, 0x2d, 0x0a, 0x5a, 0x0c, 
[1648652954.043002][268478:268478] CHIP:IN: 0x83, 
[1648652954.043041][268478:268478] CHIP:IN: Generated compressed fabric ID
[1648652954.043058][268478:268478] CHIP:IN: 0xa9, 0xc5, 0x4c, 0xf4, 0x39, 0x13, 0x38, 0xca, 
[1648652954.043323][268478:268478] CHIP:IN: Loading certs from storage
[1648652954.043399][268478:268478] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1648652954.043422][268478:268478] CHIP:IN: 0x04, 0xff, 0xaa, 0xa7, 0xd2, 0x03, 0xe6, 0x11, 
[1648652954.043438][268478:268478] CHIP:IN: 0x30, 0x79, 0xf1, 0xa7, 0x87, 0xd6, 0xb9, 0x35, 
[1648652954.043454][268478:268478] CHIP:IN: 0xb0, 0x09, 0x58, 0x98, 0xba, 0x8a, 0xa8, 0xe5, 
[1648652954.043470][268478:268478] CHIP:IN: 0xd6, 0x92, 0x74, 0xfa, 0x15, 0x43, 0x1b, 0xeb, 
[1648652954.043486][268478:268478] CHIP:IN: 0x2e, 0xbe, 0x75, 0x7e, 0x84, 0x86, 0xf2, 0xbd, 
[1648652954.043502][268478:268478] CHIP:IN: 0x48, 0x20, 0xb3, 0xa6, 0xc8, 0xbd, 0x5d, 0xba, 
[1648652954.043517][268478:268478] CHIP:IN: 0x42, 0xfc, 0x20, 0xf1, 0x15, 0x2b, 0x91, 0xfc, 
[1648652954.043533][268478:268478] CHIP:IN: 0x0f, 0x2f, 0x68, 0x7c, 0x2d, 0x0a, 0x5a, 0x0c, 
[1648652954.043547][268478:268478] CHIP:IN: 0x83, 
[1648652954.043584][268478:268478] CHIP:IN: Generated compressed fabric ID
[1648652954.043602][268478:268478] CHIP:IN: 0x58, 0xed, 0xed, 0x07, 0x6e, 0xed, 0xfe, 0x10, 
[1648652954.043930][268478:268478] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-4rRz6f)
[1648652954.044415][268478:268478] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1648652954.044463][268478:268478] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 6000 (0x1770)
[1648652954.044592][268478:268478] CHIP:ZCL: Using ZAP configuration...
[1648652954.045901][268478:268478] CHIP:DL: MDNS failed to join multicast group on wlp7s0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1648652954.046495][268478:268478] CHIP:CTL: System State Initialized...
[1648652954.046568][268478:268478] CHIP:IN: UDP::Init bind&listen port=5550
[1648652954.046618][268478:268478] CHIP:IN: UDP::Init bound to port=5550
[... 53 lines cut ...]
[1648652954.103345][268478:268478] CHIP:IN: 0xd6, 0x92, 0x74, 0xfa, 0x15, 0x43, 0x1b, 0xeb, 
[1648652954.103349][268478:268478] CHIP:IN: 0x2e, 0xbe, 0x75, 0x7e, 0x84, 0x86, 0xf2, 0xbd, 
[1648652954.103355][268478:268478] CHIP:IN: 0x48, 0x20, 0xb3, 0xa6, 0xc8, 0xbd, 0x5d, 0xba, 
[1648652954.103363][268478:268478] CHIP:IN: 0x42, 0xfc, 0x20, 0xf1, 0x15, 0x2b, 0x91, 0xfc, 
[1648652954.103372][268478:268478] CHIP:IN: 0x0f, 0x2f, 0x68, 0x7c, 0x2d, 0x0a, 0x5a, 0x0c, 
[1648652954.103379][268478:268478] CHIP:IN: 0x83, 
[1648652954.103414][268478:268478] CHIP:IN: Generated compressed fabric ID
[1648652954.103424][268478:268478] CHIP:IN: 0x58, 0xed, 0xed, 0x07, 0x6e, 0xed, 0xfe, 0x10, 
[1648652954.103442][268478:268478] CHIP:DIS: Added new fabric at index: 3, Initialized: 1
[1648652954.103447][268478:268478] CHIP:DIS: Assigned compressed fabric ID: 0x58EDED076EEDFE10, node ID: 0x000000000001B669
[1648652954.103451][268478:268478] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0x58EDED076EEDFE10
[1648652954.103478][268478:268478] CHIP:IN: UDP::Init bind&listen port=5550
[1648652954.103536][268478:268478] CHIP:IN: UDP::Init bound to port=5550
[1648652954.103546][268478:268478] CHIP:IN: UDP::Init bind&listen port=5550
[1648652954.103574][268478:268478] CHIP:IN: UDP::Init bound to port=5550
[1648652954.103589][268478:268478] CHIP:IN: TransportMgr initialized
[1648652954.103722][268478:268483] CHIP:DL: CHIP task running
[1648652954.103783][268478:268483] CHIP:DIS: DNS-SD StartServer modeHasValue=1 modeValue=0
[1648652954.105422][268478:268483] CHIP:DL: MDNS failed to join multicast group on wlp7s0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1648652954.106017][268478:268483] CHIP:DIS: CHIP minimal mDNS started advertising.
[1648652954.106496][268478:268483] CHIP:SPT: VerifyOrDie failure at ../../src/app/server/Dnssd.cpp:255: mFabricTable != nullptr
Aborted (core dumped)

Observation
start-udc-server was working as expected on TE8/rc3 tag.

cabin15, 2022-03-30:
Anyone know about this if this is regression? or additional step is required for this?
Thank you in advance.

cabin15, 2022-04-27:
Checked on aa566b02070252043015a90946fea0685103d7dd and issue is still reproducing.
It seems like no one is watching this.

woody-apple, 2022-11-02:
Issue Scrub: @cabin15 is this still an issue? Can you please re-open with more recent logs if so.

closed 2022-11-02 by woody-apple