project-chip/connectedhomeip #16837: Operational Discovery failed if  a node advertise A and AAAA records in the same time. Incorrect node address discovering.

AntonGrey, 2022-03-30:
#### Node address can't be discovered correctly if node publish both A and AAAA records. Commissioner build with disabled IPv4.
In case if commissioner build with enabled  IPv4 discovering works well.
 * expected behavior: A record should be ignored if IPv4 disabled
 * actual behavior: A record used to discover node address.
 * steps to reproduce: Pair a commissionee which advertise  both A and AAAA records and  chip-tool with disabled IPv4
 * most interesting part of a log:
 ``` 
[1648655043.939507][3664708:3664713] CHIP:TOO: Secure Pairing Success
[1648655043.939656][3664708:3664713] CHIP:DIS: Attempt to mDNS broadcast failed:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1648655043.939694][3664708:3664713] CHIP:DMG: ICR moving to [AwaitingDe]
[1648655043.939711][3664708:3664713] CHIP:EM: Sending Standalone Ack for MessageCounter:3596501 on exchange 12808i
[1648655043.939789][3664708:3664713] CHIP:IN: Prepared secure message 0x7fb32e7fb070 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 12808i with MessageCounter:6735112.
[1648655043.939810][3664708:3664713] CHIP:IN: Sending encrypted msg 0x7fb32e7fb070 with MessageCounter:6735112 to 0x0000000000000001 (1) at monotonic time: 66917013 msec
[1648655043.939850][3664708:3664713] CHIP:EM: Flushed pending ack for MessageCounter:3596501 on exchange 12808i
[1648655043.940454][3664708:3664713] CHIP:DIS: Node ID resolved for 0x0000000000000001
[1648655043.940466][3664708:3664713] CHIP:DIS:     Addr 0: [::ffff:192.168.88.11]:5540
[1648655043.940477][3664708:3664713] CHIP:DIS:     Addr 1: [fe80::217:88ff:feb0:ba3]:5540
[1648655043.940488][3664708:3664713] CHIP:CTL: OperationalDiscoveryComplete for device ID 0x0000000000000001
[1648655043.940879][3664708:3664713] CHIP:IN: Generating DestinationID. Fabric ID 0x0000000000000001, Dest node ID 0x0000000000000001, Random data
...
[1648655043.941312][3664708:3664713] CHIP:CTL: Device connection failed. Error ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1648655043.941327][3664708:3664713] CHIP:CTL: Failed to perform commissioning step 17
[1648655043.941338][3664708:3664713] CHIP:CTL: Rendezvous cleanup
[1648655043.941350][3664708:3664713] CHIP:TOO: Device commissioning Failure: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1648655043.941384][3664708:3664713] CHIP:CTL: Address resolved for node: 0x0000000000000001
[1648655043.941397][3664708:3664713] CHIP:CTL: OnNodeIdResolved was called for a device with no active sessions, ignoring it.
[1648655043.941514][3664708:3664708] CHIP:-: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable at ../examples/chip-tool/commands/common/CHIPCommand.cpp:62
[1648655043.941546][3664708:3664708] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
```
[failed_discovery.log](https://github.com/project-chip/connectedhomeip/files/8382302/failed_discovery.log)

 #### Proposed Solution
Type A records should ignore if commissioner doesn't support IPv4.
Below you can see a simple fix for this issue.

``` diff
--- a/src/lib/dnssd/Resolver_ImplMinimalMdns.cpp
+++ b/src/lib/dnssd/Resolver_ImplMinimalMdns.cpp
@@ -275,6 +275,7 @@ void PacketDataReporter::OnResource(ResourceType type, const ResourceData & data
         }
         break;
     case QType::A: {
+#if INET_CONFIG_ENABLE_IPV4
         Inet::IPAddress addr;
         if (!ParseARecord(data.GetData(), &addr))
         {
@@ -292,6 +293,7 @@ void PacketDataReporter::OnResource(ResourceType type, const ResourceData & data
                 OnDiscoveredNodeIPAddress(addr);
             }
         }
+#endif
         break;
     }
     case QType::AAAA: {
```

andy31415, 2022-04-05:
@AntonGrey could you validate that #17074 fixes your issue as well? I believe the bug is that parsing succeeds at all (it reaches into IPAddress class internals).

AntonGrey, 2022-04-05:
I confirm, #17074 fixes my issue. Thank you.

PR #17074 (2022-04-05): Fail A record parsing if IPV4 is disabled in CHIP
#### Problem

MinMDNS blindly parses A records even though code relies that IPV4 disabled means no IPV4 addresses anywhere.

#### Change overview
Refuse to parse A records if no IPV4 support.

Fixes #16837

#### Testing
Compile and unit test updated.
closed 2022-04-05 by AntonGrey