project-chip/connectedhomeip #16849: Darwin dns-sd advertising needs to use the right hostname

bzbarsky-apple, 2022-03-30:
#### Problem
Right now operational and commissionable DNS-SD advertising on Darwin uses the default hostname of the device involved, because we are passing `NULL` for the `host` argument when calling `DNSServiceRegister`.

Per spec, the hostname is supposed to be the MAC (or whatever randomized thing we are pretending is the MAC).

I tried implemeting this as follows:
1. Modify `Register` in `src/platform/Darwin/DnssdImpl.cpp` to take a `hostname` argument.
2. Pass `service->mHostName` to `Register` from `ChipDnssdPublishService` in that file.
3. Do some munging on the hostname in `Register`:
    ```
    // Treat empty hostname as null.
    if (hostname != nullptr && strcmp(hostname, "") == 0) {
        hostname = nullptr;
    }

    std::string hostnameStr;
    if (hostname != nullptr) {
        hostnameStr = std::string(hostname) + '.' + kLocalDot;
        hostname = hostnameStr.c_str();
    }
    ```
4. Pass `hostname` to `DNSServiceRegister`.

That fixes the advertisements to have the right hostname, afaict, but address resolution then fails because there is no IP associated with that hostname.  The documentation at https://developer.apple.com/documentation/dnssd/1804733-dnsserviceregister?language=objc says to call [DNSServiceRegisterRecord](https://developer.apple.com/documentation/dnssd/1804727-dnsserviceregisterrecord?language=objc) to handle that, but I could not figure out the right arguments to pass it, even when hard-coding the relevant IPs, to get past "invalid argument" errors it was giving me.

 #### Proposed Solution
* Figure out whether the combination of passing a hostname to `DNSServiceRegister` and calling `DNSServiceRegisterRecord` is actually what we want to be doing here.
* If so, sort out how to actually call `DNSServiceRegisterRecord` properly.

andy31415, 2022-06-15:
Would it be easier to update the spec to recommend using the mac as hostname?
Or does that potentially cause issues because macs (or other devices) may have overly long hostnames that devices then fail to work with due to embedded constrains?

bzbarsky-apple, 2022-06-16:
> Or does that potentially cause issues because macs (or other devices) may have overly long hostnames that devices then fail to work with due to embedded constrains?

Exactly.  We ran into this because minmdns was using a buffer sized to the spec thing and failing out if the hostname did not
fit in there, iir.

PR #21985 (2022-08-17): Register the DNS service for the accessory with a valid hostname
- Call DNSServiceRegisterRecord to register a record for the hostname for all interfaces
  before calling DNSServiceRegister.

#### Problem
What is being fixed?  Examples:
* Fixes #16849 

#### Change overview
Add support to register DNS service with a valid hostname that is made of a string containing MAC address and domain name.

#### Testing
How was this tested? (at least one bullet point required)
* Tested using mac all-clusters-app and chip-tool. Accessory is advertising a service with the desired hostname and we are able to [... cut]
PR #22010 (2022-09-12): Register the DNS service for the accessory with a valid hostname
- Call DNSServiceRegisterRecord to register a record for the hostname for all interfaces
  before calling DNSServiceRegister.

#### Problem
What is being fixed?  Examples:
* Fixes #16849

#### Change overview
Register a DNS service with a valid hostname

#### Testing
How was this tested? (at least one bullet point required)
*Tested using mac all-clusters-app and chip-tool
closed 2022-09-12 by woody-apple