project-chip/connectedhomeip #16866: [SQA] Not able to build out folder.

Survensa, 2022-03-31:
Issue :  In CHIP TOOL we are not able to build OUT folder 

Command Used : `ninja -C out/debug/`
 
 Error Log : 

> ubuntu@ubuntu:~/connectedhomeip/examples/chip-tool$ ninja -C out/debug/
ninja: Entering directory `out/debug/'
[153/581] c++ obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o
FAILED: obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o 
aarch64-linux-gnu-g++ -MMD -MF obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -Wconversion -D_REENTRANT -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -Wno-implicit-fallthrough -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCONFIG_USE_SEPARATE_EVENTLOOP=true -DCONFIG_USE_INTERACTIVE_MODE=true -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -I../.. -I../../third_party/connectedhomeip/zzz_generated/chip-tool -I../../third_party/connectedhomeip/src/lib -I../../third_party/connectedhomeip/src/include -I../../third_party/connectedhomeip/src -Igen/include -I../../third_party/connectedhomeip/zzz_generated/app-common -I../../third_party/connectedhomeip/examples/chip-tool/include -I../../third_party/connectedhomeip/config/standalone -I../../third_party/connectedhomeip/third_party/nlassert/repo/include -I../../third_party/connectedhomeip/third_party/nlio/repo/include -I../../third_party/connectedhomeip/third_party/mbedtls/repo/include -I../../third_party/connectedhomeip/third_party/inipp/repo/inipp -I../../third_party/connectedhomeip/zzz_generated -I../../third_party/connectedhomeip/zzz_generated/controller-clusters -I../../third_party/connectedhomeip/third_party/jsoncpp/repo/include -I../../third_party/connectedhomeip/examples/common/tracing -c ../../commands/interactive/InteractiveCommands.cpp -o obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o
../../commands/interactive/InteractiveCommands.cpp:21:10: fatal error: readline/history.h: No such file or directory
   21 | #include <readline/history.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
[158/581] c++ obj/chip-tool.main.cpp.o
ninja: build stopped: subcommand failed.

bpetersen-grundfos, 2022-03-31:
Have same issue on PI, think its related to this pull:
https://github.com/project-chip/connectedhomeip/pull/16135
If you set config_use_interactive_mode = false in BUILD.gn chip-tool builds, but not sure of the consequences.

vivien-apple, 2022-03-31:
> Have same issue on PI, think its related to this pull: #16135 If you set config_use_interactive_mode = false in BUILD.gn chip-tool builds, but not sure of the consequences.
> 

The only consequence should be that `chip-tool interactive start` will not work anymore

> Or you can install readline lib: apt-get install libreadline-dev

+1.

Survensa, 2022-04-04:
Still facing the same issue but able to build only using above workaround.

Commit Used : 41a431dcc7c123699818636afd01b57412a6f3a4

Workaround used to build : `sudo apt-get install libreadline-dev`

bzbarsky-apple, 2022-04-04:
It sounds like we need to add the bit about libreadline-dev to our "how to set up a build environment" documentation?  @vivien-apple

Survensa, 2022-04-19:
The issue has been resolved in latest SHA

closed 2022-04-19 by Survensa