project-chip/connectedhomeip #16897: Since commit 11ab5a3 of Example/Chip-Tool, Ninja Compile for Example/Chip-Tool failed

RudyGOGOGO, 2022-03-31:
After this commit https://github.com/project-chip/connectedhomeip/commit/11ab5a3c163b677894b912960d1e9252a17aa6a9, when I ran the following script, I got the error "fatal error: readline/history.h: No such file or directory" which did not happen before commit 11ab5a3 of Example/Chip-Tool "Chip tool interactive(#16135)"
`scripts/examples/gn_build_example.sh examples/chip-tool out/` suggedted in https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool

Please check the attached file with all details about the error above
[errorLogsAboutNinjaCompileFailed.txt](https://github.com/project-chip/connectedhomeip/files/8392858/errorLogsAboutNinjaCompileFailed.txt)

RudyGOGOGO, 2022-04-01:
looks like it happened because the new dependency is introduced `libreadline-dev`
After I installed it, it works for me.
It would be better to update the **Installing prerequisites on Linux** in https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

andy31415, 2022-04-05:
This needs fixing in the vscode image: we need chip-build-vscode to be able to compile all builds out of the box, right now it does not:

```
2022-04-02 13:13:25 INFO    FAILED: obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o 
2022-04-02 13:13:25 INFO    clang++ -MMD -MF obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o.d --target=aarch64-linux-gnu -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough …-tool/third_party/connectedhomeip/third_party/jsoncpp/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/examples/common/tracing -c ../../examples/chip-tool/commands/interactive/InteractiveCommands.cpp -o obj/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o
2022-04-02 13:13:25 INFO    ../../examples/chip-tool/commands/interactive/InteractiveCommands.cpp:21:10: fatal error: 'readline/history.h' file not found
2022-04-02 13:13:25 INFO    #include <readline/history.h>
```

andy31415, 2022-04-05:
It looks like this is related to ARM specifically - our images compile for x64 but not for arm build cross-compiles.

I will for now add a noglob and will see about what needs to be done to add libreadline to the arm sysroot.

andy31415, 2022-04-05:
created #17084 for the glob bit. The build requirements and arm sysroot will have to be done separately (hoping to get time soon ... but it is somewhat tedious).

RudyGOGOGO, 2022-04-06:
@andy31415 I see you mentioned the cross-compile in your comment.
I am working on cross compiling the [ChipTool](https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool) for OpenWrt, I get the OpenWrt Cross Compiling Environment ready, however, I could not find an easy way to modify the [BUILD.GN](https://github.com/project-chip/connectedhomeip/blob/master/examples/chip-tool/BUILD.gn) in order to use OpenWrt Cross Compiler as well as all other dependencies like headers

Would you mind providing some hint or instructions about how to cross compile the ChipTool for OpenWrt?

closed 2022-10-12 by stale[bot]

xlfdan, 2023-08-04:
@RudyGOGOGO Have you find a way to cross compile the ChipTool for OpenWrt? Thank you.