project-chip/connectedhomeip #16938: [build] submodule pointing to stale commit

turon, 2022-04-01:
#### Problem
```
fatal: Fetched in submodule path 'third_party/efr32_sdk/wiseconnect-wifi-bt-sdk', but it did not contain 30361d9e5e70d42b2dba4da27aa65cff4481ac18. Direct fetching of that commit failed.
```

 #### Proposed Solution
Fix  `third_party/efr32_sdk/wiseconnect-wifi-bt-sdk` to point to a valid commit.

turon, 2022-05-05:
@jepenven-silabs is someone from SiLabs able to look into this?

turon, 2022-05-05:
This submodule is blocking build on a fresh clone of the tree on macos:

```
$ ./gn_build.sh 
.--------------------------------
-- Environment bringup
'--------------------------------
Submodule path 'examples/common/QRCode/repo': checked out '8518684c0f33d004fa93971be2c6a8eca3167d1e'
Submodule path 'examples/common/m5stack-tft/repo': checked out 'a1ece2c500cf179db8f0c7dbc7bb8f3ace562142'
Submodule path 'third_party/cirque/repo': checked out '262ea7724667229f0d21d6a1c1e96d90415f0906'
Submodule path 'third_party/efr32_sdk/repo': checked out '175cbe761d2fab133725dab01abb1a149e23eb20'
error: Your local changes to the following files would be overwritten by checkout:
	examples/at_commands/teraterm/bt_spp_slave/resources/receive-message.png
Please commit your changes or stash them before you switch branches.
Aborting
fatal: Unable to checkout '4491826e48bfaaa6ce4d42ad5769d7f6447b8cbb' in submodule path 'third_party/efr32_sdk/wiseconnect-wifi-bt-sdk'
mturon-macbookpro1:turon mturon$ ./gn_build.sh 
.--------------------------------
-- Environment bringup
'--------------------------------
error: Your local changes to the following files would be overwritten by checkout:
	examples/at_commands/teraterm/bt_spp_slave/resources/receive-message.png
Please commit your changes or stash them before you switch branches.
Aborting
fatal: Unable to checkout '4491826e48bfaaa6ce4d42ad5769d7f6447b8cbb' in submodule path 'third_party/efr32_sdk/wiseconnect-wifi-bt-sdk'
```

tewarid, 2022-05-09:
Commit 60adc7e7afb925fd55256c2baaa4ca9663a65d17 seems to have fixed this issue for me.