project-chip/connectedhomeip #16954: Every alternate execution of Toggle on endpoint2 throws an error 0x86

phonnakasturi-apple, 2022-04-01:
On CHIPTool app, when trying to test On Off cluster;
On/Off/Toggle on endpoint2 throws an error 0x86 on endpoint 2

Steps to reproduce;

1. Pair M5 board with CHIPTool app
2. Select Light On/Off cluster
3. Update #of light endpoints: 2
4. On Light (endpoint 2) -> First Click on Toggle -> works fine (Toggle command success)
5. On Light (endpoint 2) -> Second Click on Toggle -> fails (An error occurred 0x86)

phonnakasturi-apple, 2022-04-02:
Attaching M5 device logs along with iOS log
[M5 logs.txt](https://github.com/project-chip/connectedhomeip/files/8401569/M5.logs.txt)
[iOS logs.txt](https://github.com/project-chip/connectedhomeip/files/8401570/iOS.logs.txt)
s

bzbarsky-apple, 2022-04-03:
0x86 is UNSUPPORTED_ATTRIBUTE.

The m5 log says:
```
I (197201) chip[ZCL]: ERR: reading onMode 86
```
which is what happens when `EMBER_AF_PLUGIN_MODE_SELECT` is defined (which is global state) but the `OnMode` attribute is not supported on the endpoint (which is endpoint-specific state).  Which is what happens with endpoint 2 of all-clusters-app: it supports the on/off cluster but _not_ the mode select cluster.

Looks like fallout from #15888.  This code needs to be at the very least checking that the mode select cluster exists on the endpoint before trying to get its attributes (much less failing out on failure to get them).

bzbarsky-apple, 2022-04-03:
And we should add YAML tests for this, using endpoint 2 of the all-clusters-app.

du48s03, 2022-04-03:
This PR already fixes this problem: https://github.com/project-chip/connectedhomeip/pull/16791.

bzbarsky-apple, 2022-04-04:
Yep, confirmed.  Thanks!

bzbarsky-apple, 2022-04-04:
Though it would still be good to add a test @du48s03

closed 2022-04-04 by bzbarsky-apple