project-chip/connectedhomeip #16961: CHTPTool cli does not complie on M1 Mac ->  ld: symbol(s) not found for architecture x86_64

phonnakasturi-apple, 2022-04-02:
CHTPTool cli does not complie on M1 Mac but works on intel based Mac

SHA: f7339952cb05918cb811ced511cd6ff909646425

Note: It works on intel based macs

```
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[1388/1558] ld tests/TestPlatformMgr
FAILED: tests/TestPlatformMgr tests/TestPlatformMgr.map 
g++ -target x86_64-apple-macos10.15 -O0 -fPIC -Werror -fdiagnostics-color -L/opt/homebrew/Cellar/openssl@3/3.0.2/lib -Wl,-map,tests/TestPlatformMgr.map -Wl,-dead_strip @tests/TestPlatformMgr.rsp -o tests/TestPlatformMgr
ld: warning: ignoring file /opt/homebrew/Cellar/openssl@3/3.0.2/lib/libssl.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/Cellar/openssl@3/3.0.2/lib/libcrypto.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "_RAND_priv_bytes", referenced from:
      chip::Crypto::DRBG_get_bytes(unsigned char*, unsigned long) in libChipCrypto.a(libChipCrypto.CHIPCryptoPALOpenSSL.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[1389/1558] ld tests/TestTlvToJson
ld: warning: ignoring file /opt/homebrew/Cellar/openssl@3/3.0.2/lib/libssl.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/Cellar/openssl@3/3.0.2/lib/libcrypto.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
[1390/1558] ld tests/TestBytesCircularBuffer
ld: warning: ignoring file /opt/homebrew/Cellar/openssl@3/3.0.2/lib/libssl.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/Cellar/openssl@3/3.0.2/lib/libcrypto.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
```

ReneJosefsen, 2022-04-04:
Seems to be related to https://github.com/project-chip/connectedhomeip/issues/16759

ReneJosefsen, 2022-04-04:
Seems to be an issue with gn, I have posted some details on the other issue.

kean-apple, 2022-04-07:
This works when using gn gen out/debug --args='chip_mdns="platform" chip_crypto="mbedtls" chip_inet_config_enable_ipv4=false'

closed 2022-04-07 by kean-apple