project-chip/connectedhomeip #17049: Occupancy sensing cluster [ TC-OCC-3.1,][TC-OCC-3.2] - Test step shows operate dut manually to change the occupancy status

somu1710, 2022-04-05:
Problem
In occupancy sensing cluster testcase [ TC-OCC-3.1] and [TC-OCC-3.2] -  we have to manually change the occupancy status,currently there is no provision to enter this value or to pre-configure this value. If there is a provision to con=figure this value, it will be helpful to verify this feature.For TC-3.1& 3.2 - Row no.3  

Here I attached the link for TE 9 Verification steps
https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit#gid=834091041

[image]
[image]

closed 2022-04-22 by somu1710

s-mcclain, 2022-07-26:
Hey @somu1710 can I ask why you closed out this issue? Seems like this is still a problem which makes the tests TC-OCC-3.2 impossible to execute with the test harness at the moment

somu1710, 2022-07-27:
@s-mcclain  Actually I closed this ticket because of manual operation we cant perform by using the Raspi device ,App- All clusters app.so this can be executable on real DUT .So we thought of  no need this ticket and we closed all manual operation required test cases.Should I need to reopen the issue again?