project-chip/connectedhomeip #17053: Receiving unwanted error message in multi-admin of ECM test-cases

sowmyassp, 2022-04-05:
Issue: Receiving unwanted error message in multi-admin of ECM test-cases "Commissioning discovery over BLE failed: ../../third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:830: CHIP Error 0x00000032: Timeout
"When we tried to commissioning TH2 to DUT after successful commissioning TH1 to DUT we are facing above issue which is unnecessary because all the commissioning flow is doe and also got message "Device commissioning completed with success".

Steps to repoduce : 
DUT side : sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app --wifi

TH side : ./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840 (on TH1)
./chip-tool pairing open-commissioning-window 1 1 100 1000 3840(on TH1)
./chip-tool pairing manualcode 2 35484132896(on TH2)

Issue screenshot : 
[image]

PFA logs:
[ECM_fail.txt](https://github.com/project-chip/connectedhomeip/files/8418244/ECM_fail.txt)
[ECM_fail_dut.txt](https://github.com/project-chip/connectedhomeip/files/8418246/ECM_fail_dut.txt)

bzbarsky-apple, 2022-04-05:
But fundamentally, I expect the issue is that `SetUpCodePairer::StopConnectOverBle` gets called, which does:
```
    return mBleLayer->CancelBleIncompleteConnection();
```
That lands in `BleLayer::CancelBleIncompleteConnection` which does:
```
    CHIP_ERROR err = mConnectionDelegate->CancelConnection();
    if (err == CHIP_ERROR_NOT_IMPLEMENTED)
    {
        ChipLogError(Ble, "BleConnectionDelegate::CancelConnection is not implemented.");
    }
```
and based on my assumption baed on the log above that the commissioner is running on Linux, that lands in the Linux `BLEManagerImpl::CancelConnection` which does:
```
CHIP_ERROR BLEManagerImpl::CancelConnection()
{
    return CHIP_ERROR_NOT_IMPLEMENTED;
}
```

So first question.  Is there a "BleConnectionDelegate::CancelConnection is not implemented." error log much earlier in the log.  @sowmyassp

sowmyassp, 2022-04-06:
Commit used is - 41a431dcc7c123699818636afd01b57412a6f3a4 
For the above question yes we get "BleConnectionDelegate::CancelConnection is not implemented." error log much earlier in the log.

bzbarsky-apple, 2022-04-06:
Alright, that's the issue then: Linux does not implement canceling the BLE connection once we know we don't need it anymore, and eventually it times out and logs an error on timeout....

sowmyassp, 2022-04-06:
Then can we say this test-case is pass or its an major issue??

bzbarsky-apple, 2022-04-06:
Testcase is passing as long as the device ends up commissioned and the client exists with a success status, I would think.  The logging along the way doesn't really matter.

closed 2022-04-13 by sowmyassp