project-chip/connectedhomeip #17088: We should move DeviceInfoProviderImpl away from PlatformManagerImpl and make Server have a reference to it

yufengwangca, 2022-04-06:
#### Problem
DeviceInfoProviderImpl is used in a way that is very hidden. I didn't even know it existed until today. This is not a platform item, it is a per-product item. We should move this away from PlatformManagerImpl and make Server have a reference to it, so that eventually we can init the UserLabel clutser to use the DeviceInfoProvider rather than these deep hidden singletons

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-05-21:
Can't actually implement a compliant fixed label cluster on top of what we have right now....

woody-apple, 2022-05-26:
Spec Review: This is just an implementation problem. Closing, as we also believe this is complete

closed 2022-05-26 by woody-apple