project-chip/connectedhomeip #17106: Networks attribute in Network commissioning has a list without connected status as TRUE even after commissioning

Sivaramgrl, 2022-04-06:
#### Problem
Networks attribute in Network commissioning has a list without connected status as TRUE even after commissioning

###  expected behaviour
After the commissioning the Networks attribute list need to have one entry with Connected status as TRUE

###  actual behaviour
After the commissioning the Networks attribute list has no entry with connected status as TRUE

### Logs:
 >ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning read networks 1 0

[1649249212.072823][30681:30686] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 2702518944
[1649249212.072912][30681:30686] CHIP:TOO:   Networks: 1 entries
[1649249212.073027][30681:30686] CHIP:TOO:     [1]: {
[1649249212.073067][30681:30686] CHIP:TOO:       NetworkID: 47524C507269766174655F455854
[1649249212.073101][30681:30686] CHIP:TOO:       Connected: FALSE
[1649249212.073135][30681:30686] CHIP:TOO:      }
[1649249212.073266][30681:30686] CHIP:EM: Sending Standalone Ack for MessageCounter:8988360 on exchange 58610i

 #### Proposed Solution
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning read networks 1 0

[1649249212.072823][30681:30686] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 2702518944
[1649249212.072912][30681:30686] CHIP:TOO:   Networks: 1 entries
[1649249212.073027][30681:30686] CHIP:TOO:     [1]: {
[1649249212.073067][30681:30686] CHIP:TOO:       NetworkID: 47524C507269766174655F455854
[1649249212.073101][30681:30686] CHIP:TOO:       Connected: TRUE
[1649249212.073135][30681:30686] CHIP:TOO:      }
[1649249212.073266][30681:30686] CHIP:EM: Sending Standalone Ack for MessageCounter:8988360 on exchange 58610i

[networks_issue.txt](https://github.com/project-chip/connectedhomeip/files/8427057/networks_issue.txt)

Sivaramgrl, 2022-04-06:
FYI, the InterfaceEnaled attribute has a value TRUE.

### LOGS:
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning read interface-enabled 1 0

[1649250241.092686][30726:30731] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0004 DataVersion: 2702518944
[1649250241.092744][30726:30731] CHIP:TOO:   InterfaceEnabled: TRUE
[1649250241.092852][30726:30731] CHIP:EM: Sending Standalone Ack for MessageCounter:3120276 on exchange 4316i

erjiaqing, 2022-04-15:
From my comment in 15782

https://github.com/project-chip/connectedhomeip/issues/15782#issuecomment-1098648526

```
In [15]: await devCtrl.SendCommand(1, 0, Clusters.GeneralCommissioning.Commands.ArmFailSafe(expiryLengthSeconds=360))
Out[15]: 
ArmFailSafeResponse(
│   errorCode=<CommissioningError.kOk: 0>,
│   debugText=''
)

In [16]: await devCtrl.SendCommand(1, 0, Clusters.NetworkCommissioning.Commands.AddOrUpdateWiFiNetwork(ssid=b'(my home wifi ssid)', credentials=b'(my home wifi password)'))
Out[16]: 
NetworkConfigResponse(
│   networkingStatus=<NetworkCommissioningStatus.kSuccess: 0>,
│   debugText=None,
│   networkIndex=0
)

In [19]: await devCtrl.ReadAttribute(1, [(0, Clusters.NetworkCommissioning)], returnClusterObject=True)
Out[19]: 
{
│   0: {
│   │   <class 'chip.clusters.Objects.NetworkCommissioning'>: NetworkCommissioning(
│   │   │   (data version)=3994585040,
│   │   │   maxNetworks=1,
│   │   │   networks=[
│   │   │   │   NetworkInfo(
│   │   │   │   │   networkID=b'(my home wifi ssid)',
│   │   │   │   │   connected=True
│   │   │   │   )
│   │   │   ],
```

I found connected is True here. Do you have a full steps for reproducing this?

Based our code related to this, we set the "connected" flag by checking if the current connected network is the one we commissioned, so could you use some other tools (like network manager) to confirm that the actual network the RPi is connected to?

Sivaramgrl, 2022-04-19:
Thanks for sharing the input @erjiaqing,
Based on your comments, I worked on the Networks commissioning cluster commands
Below is my observation:

Used Commit id:
commit https://github.com/project-chip/connectedhomeip/commit/87938704ed93dde93e960605ae6a714923efa2e0
Date: Mon Apr 18 18:59:53 2022 -0700

Steps followed:
before the failsafe timeout, I send the add or update network request message with NetworkID and with the same NetworkID I sent the connect-network request message too. The DUT responds with success for both commands.

Output:
The sequence was executed before failsafe timeout. But still, I could not able to find the expected output.
Confirmed the output by reading the Networks attribute, interface-enabled attribute and last-networking-status. The list holds the NetworkID which is requested in add or update Network command. But still, the list is updated as the connected status is FALSE and the interface-enabled attribute value is TRUE.

Logs:
[Full DUT report-NC.txt](https://github.com/project-chip/connectedhomeip/files/8511615/Full.DUT.report-NC.txt)
[Full TH report -NC.txt](https://github.com/project-chip/connectedhomeip/files/8511623/Full.TH.report.-NC.txt)

andy31415, 2022-04-19:
@Sivaramgrl - what platform is the DUT here?

I would make a difference here between RPI (linux - full platform) and any other actual embedded devices. Trying to asses severity - if linux does not report connected it does not a v1 blocking issue.

Survensa, 2022-04-20:
@Sivaramgrl Please close this issue 

Verified the with the commit : 20e1a4a705770d5222a8ebf04e12ab2ce0304df6

Used the cmd : `./chip-tool networkcommissioning read networks 1 0`

Got the expected outcome : 
```
[1650442167.425528][6870:6876] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 533595784
[1650442167.425636][6870:6876] CHIP:TOO:   Networks: 1 entries
[1650442167.425806][6870:6876] CHIP:TOO:     [1]: {
[1650442167.425850][6870:6876] CHIP:TOO:       NetworkID: 63686970736574757033
[1650442167.425889][6870:6876] CHIP:TOO:       Connected: TRUE
[1650442167.425927][6870:6876] CHIP:TOO:      }
[1650442167.426073][6870:6876] CHIP:EM: Sending Standalone Ack for MessageCounter:10615825 on exchange 24656i
```
Full log :
[TC_NC-4.1.txt](https://github.com/project-chip/connectedhomeip/files/8519871/TC_NC-4.1.txt)

PR #17531 (2022-04-20): [Linux] Fix some behavior of linux wifi driver
#### Problem
- Fixes #17106 - ConnectNetwork returns result too early -- before actually connected to the network
- ScanNetwork is not filtering the discovered network with interested ssids

#### Change overview
Fixed the behavior above.

#### Testing

Scan network
```
In [4]: await devCtrl.SendCommand(1, 0, Clusters.NetworkCommissioning.Commands.ScanNetworks(ssid=b'(personal ap)'))
Out[4]: 
ScanNetworksResponse(
│   networkingStatus=<NetworkCommissioningStatus.kSuccess: 0>,
│   debugText=None,
│   wiFiScanResults=[
│   │   WiFiInterfaceScanResult [... cut]
closed 2022-04-20 by Sivaramgrl