project-chip/connectedhomeip #17112: Pump Control and Configuration ControlMode and OperationMode should have min/max specified in XML

bzbarsky-apple, 2022-04-06:
#### Problem
We don't do range checks on sets to these attributes.

 #### Proposed Solution
We should be doing range checks, but that requires the valid range to be specified in the XML.

@tlykkeberg-grundfos @mgarb1

tlykkeberg-grundfos, 2022-04-07:
Hi @bzbarsky-apple I have been looking into this on ToT and I find the following in the `pump-configuration-and-control.cluster.xml` file:

```xml
    <attribute side="server" code="0x0020" define="OPERATION_MODE" type="ENUM8" min="0x00" max="0xFE" writable="true" default="0x00" optional="false">OperationMode</attribute>
    <attribute side="server" code="0x0021" define="CONTROL_MODE" type="ENUM8" min="0x00" max="0xFE" writable="true" default="0x00" optional="true">ControlMode</attribute>
```
I'm not quite sure what you mean in this context? The `min` and `max` values are specified in the attribute definition in the XML.

bzbarsky-apple, 2022-04-07:
Yes, but they are specified for the full (nullable) uint8 range, not the actual valid range for the relevant enums.

tlykkeberg-grundfos, 2022-04-07:
Fixed in #17181

tlykkeberg-grundfos, 2022-04-08:
Hi @bzbarsky-apple since the pr #17181 I have introduced this into these two attributes in the XML:
```xml
<attribute side="server" code="0x0020" define="OPERATION_MODE" type="PumpOperationMode" min="0x00" max="0x03" writable="true" default="0x00" optional="false">
      <description>OperationMode</description>
      <access op="read" privilege="view"/>
      <access op="write" privilege="manage"/>
    </attribute>
    <attribute side="server" code="0x0021" define="CONTROL_MODE" type="PumpControlMode" min="0x00" max="0x07" writable="true" default="0x00" optional="true">
      <description>ControlMode</description>
      <access op="read" privilege="view"/>
      <access op="write" privilege="manage"/>
    </attribute>
```
Here there are min/max values and the exact enum type (instead of ENUM8). Would that be sufficient or do you see any further implementation needed for this issue to be closed?

bzbarsky-apple, 2022-04-08:
#17181 fixes this, yes.  Thank you!

PR #17181 (2022-04-20): [XML] Correct the PCC cluster xml to reflect the specification
#### Problem
* The PCC cluster xml does not fully reflect the specification
* Min and max values for some attributes in the PCC was wrong
* Add some non-default access priviledges to the PCC
* Set the ClusterRevision to be '3'
* Fixes #16550
* Fixes #17112
* Fixes #9769

#### Change overview
* Added `isNullable="true"` to the attributes which is specified to have this posibillity
* Corrected the `min` and `max` for all attributes to reflect the specification
* Added RW VM to the LifeTimeRunningHours and LifeTimeConsumedEnergy attri [... cut]
closed 2022-04-20 by andy31415