project-chip/connectedhomeip #17151: TC-IDM-6.4 chip tool Darwin testing: Subscribe-Events support needed to perform the steps

Garik098, 2022-04-06:
As per step 1 description in TC-IDM-6.4, chip-tool-darwin should have support to do subscribe event.
[image]

Below is the example for basic cluster for chip-tool vs chip-tool-darwin. The subscribe event commands are needed to execute this step. Currently in chip-tool-darwin there is no support for subscribe events in any of the cluster.
[image]

bzbarsky-apple, 2022-05-28:
Note that you can use `darwin-framework-tool any subscribe-all-events` to subscribe to all events.  The Darwin framework does not currently support individual event subscriptions.

bzbarsky-apple, 2022-05-28:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/18387

closed 2022-05-28 by bzbarsky-apple