project-chip/connectedhomeip #17164: Pairing of ESP32 with chip-tool on linux platform is failing

sowmyassp, 2022-04-07:
-  Issue : 

           Pairing ESP32 with chip-tool failing on Linux platform but successfully pairing on MAC

- Expected behaviour : 

         Device commissioning completed with success

- Actual behaviour : 

          Device connection failed. Error ../../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:176: CHIP Error 0x00000032: Timeout

- Screenshot : 

[image]

- PFA logs:

[ESP.txt](https://github.com/project-chip/connectedhomeip/files/8441987/ESP.txt)
[ESP_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8441988/ESP_cntrl.txt)

sowmyassp, 2022-04-13:
Same issue facing with commitID (fa5561c23d8c1f9464da8669931affbae0660f51)

alantsai0821, 2022-04-15:
Is there anyone know how to fix this?
I just meet the same error when try to pairing EFR32MG12 lighting-app with chip-tool on ubuntu...

[EFR32-chip-tool-log.txt](https://github.com/project-chip/connectedhomeip/files/8493662/EFR32-chip-tool-log.txt)

dhrishi, 2022-04-21:
@sowmyassp Please confirm if it is working with the latest codebase on ESP32. I can successfully perform commissioning and cluster control using Linux chip-tool on latest head (4b4e035e15eab106e33a88eac6ffbe557d82e584)

sowmyassp, 2022-04-21:
Yes @dhrishi Its working with latest head

closed 2022-04-21 by sowmyassp