project-chip/connectedhomeip #17167: [TC-MF-1.11]  After 11th attempt of invalid passcode we are not geeting any error logs

sowmyassp, 2022-04-07:
Issue : 
According to test-plan after 20 unsuccessful attempts, the 21st shows that you can’t begin PASE (failure at PBKDFRequest, not obtaining a PBKDFResponse) in step 3 of [TC-MF-1.11] test-case, but here after 11  unsuccessful attempts we are not getting any error logs.

PFA logs : 
[TC_MF_1.11_cntrl02.txt](https://github.com/project-chip/connectedhomeip/files/8442375/TC_MF_1.11_cntrl02.txt)

sowmyassp, 2022-04-07:
For [TC_MF_1.12] After 4th attempt of invalid passcode we are not getting any error log

PFA logs : 
[Uploading TC_MF_1.12.txt…]()

sowmyassp, 2022-04-19:
@bzbarsky-apple Can you please take a look at this issue

bzbarsky-apple, 2022-04-19:
I am seeing this fail after 15 failed attempts, but definitely seeing it fail.  What happens is that in `CommissioningWindowManager::HandleFailedAttempt` the call to `AdvertiseAndListenForPASE` fails.

And that fails because we fail to allocate a session in `PairingSession::AllocateSecureSession`.  And looking at the session pool:

1. It's a `BitmapObjectPool`, so is actually fixed-size.
2. The pool size is `CHIP_CONFIG_PEER_CONNECTION_POOL_SIZE`, which is 16.

So we seem to be leaking sessions, at first glance...

bzbarsky-apple, 2022-04-19:
Ah, this is because we have started allocating sessions up front, before we know pairing has been successful, and never clean them up afterward. Fix coming up.

closed 2022-04-20 by bzbarsky-apple