project-chip/connectedhomeip #17193: pw: command not found

donny681, 2022-04-08:
#### Problem
When  source ./scripts/bootstrap.sh,it shows that pw: command not found
 
`  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

   Downloading and installing packages into local source directory:

    Setting up CIPD package manager...done (3.0s)
    Setting up Python environment.....done (5.7s)
    Setting up pw packages............skipped (0.1s)
    Setting up Host tools.............done (0.1s)
    Activating environment (setting environment variables):
    Setting environment variables for CIPD package manager...done
    Setting environment variables for Python environment.....done
    Setting environment variables for pw packages............skipped
    Setting environment variables for Host tools.............done
    Checking the environment:
    pw: command not found
    Error during bootstrap--see messages above.
`

TheWalkingDead1024, 2022-04-25:
> #### Problem
> When source ./scripts/bootstrap.sh,it shows that pw: command not found
>
> ` BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
>
> Downloading and installing packages into local [...]

I have met the same issue. Have you fixed it?

coderWu1990, 2022-04-25:
Delete .environment dir,and retry,will be ok

coderWu1990, 2022-04-25:
> Delete .environment dir,and retry,will be ok

tewarid, 2022-08-19:
> Delete .environment dir,and retry,will be ok

Did not fix the error on macOS 13.0 Beta (22A5321d).

tewarid, 2022-08-19:
> > Delete .environment dir,and retry,will be ok
> 
> Did not fix the error on macOS 13.0 Beta (22A5321d).

Tried with a clean repo per #10766 and it works now.

RedPoppies, 2022-10-30:
Delete .environment dir,and retry.
This problem still exists.

samleemc, 2023-02-26:
same issue, still stuck

JonathanWitthoeft, 2023-10-18:
still exists. Suggestions above of deleting  .environment dir a nd trying again works

otamrll09, 2023-12-05:
I had the same issue here.
Suggestions above of deleting .environment dir and trying again works²

Environment:
Ubuntu 22.04.3 LTS - Virtual Machine
Project CHIP Version: 1.2.0.1