project-chip/connectedhomeip #17203: [SQA] TC-NC-2.5 & TC-NC-2.6 Current commissioning flow doesnt allow to verify error scenarios

Survensa, 2022-04-08:
Issue : The command need to be executed before commissioning which is not possiable. 

[image]

Survensa, 2022-04-08:
@Sivaramgrl Please look into it and add some comments.

erjiaqing, 2022-04-14:
Why not possible?

I received failure response before issue the ArmFailSafe command. You should be able to send the command without a commissioning flow.

Sivaramgrl, 2022-04-19:
@erjiaqing & @Survensa,

### With the commit id : 
commit 87938704ed93dde93e960605ae6a714923efa2e0
Date:   Mon Apr 18 18:59:53 2022 -0700
Now we can verify these test cases [TC-NC-2.5 & TC-NC-2.6] which are changed as [TC-DM-4.5 & TC-DM-4.6]

### Observation: 
I can see the DUT responds with UNSUPPORTED_ACCESS when the Network commissioning cluster commands are sent after the failsafe timeout. 

### Logs: 
[2.5_TH test case-NC( WiFi).txt](https://github.com/project-chip/connectedhomeip/files/8512265/2.5_TH.test.case-NC.WiFi.txt)
[2.6_DUT test case-NC(Thread).txt](https://github.com/project-chip/connectedhomeip/files/8512266/2.6_DUT.test.case-NC.Thread.txt)
[2.6_TH test case-NC(Thread).txt](https://github.com/project-chip/connectedhomeip/files/8512267/2.6_TH.test.case-NC.Thread.txt)
[2.5_DUT test case-NC( WiFi).txt](https://github.com/project-chip/connectedhomeip/files/8512268/2.5_DUT.test.case-NC.WiFi.txt)

closed 2022-04-19 by Survensa