project-chip/connectedhomeip #17205: [TC-DD-3.12] - Step 2b says DUT should terminate the commissioning process where instead commissioning process is done successfully

sowmyassp, 2022-04-08:
Issue : According to test-plan of [TC-DD-3.12] Step 2b says DUT should terminate the commissioning process where instead commissioning process is done successfully.

Test-plan reference : 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#tc-dd-3-12-commissioning-flow-qr-code-negative-scenario-dut-commissioner

Screenshot : 
[image]
  
PFA logs : 
[DD_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8451618/DD_cntrl.txt)
[DD_dut.txt](https://github.com/project-chip/connectedhomeip/files/8451619/DD_dut.txt)

sowmyassp, 2022-04-13:
Similarly Step 4d and 4f says DUT should terminate the commissioning process when we pass invalid QR Code where instead commissioning process is done successfully

erjiaqing, 2022-05-09:
Hi, from your logs posted, I only found logs for discriminator 3841, could you provide another logs for the case of discriminator = 3844?

erjiaqing, 2022-05-09:
After checking the spec again, all manual pairing codes contains only 4 MSB of the discriminator.

sowmyassp, 2022-05-10:
Here is the latest log for commit ID (11587f5bf0bb9e621edd4c867c0bc942d804d1ce) 
[DD-3.14_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8660310/DD-3.14_cntrl.txt)
[DD-3.14_dut.txt](https://github.com/project-chip/connectedhomeip/files/8660312/DD-3.14_dut.txt)

**Steps to reproduce** :

- DUT side :

1. sudo rm -rf /tmp/chip_*
2. sudo ./chip-all-clusters-app --wifi (Discriminator - 3840, Passcode - 20202021)

**QR code received : MT:-24J0YXE00KA0648G00**

- TH side

1. ./chip-tool payload generate-manualcode --discriminator 3840 --setup-pin-code 20202021 --version 001 --vendor-id 65521 --product-id 0x8001 --commissioning-mode 1

**QR Code: MT:.24J029Q00KA0648G00**

2. ./chip-tool pairing qrcode 1 MT:.24J029Q00KA0648G00

cjandhyala, 2022-05-13:
@sowmyassp, as we discussed the other day this issue i.e. generating the same QR code, is on the chip-tool side, not the commissioner issue. Pls close this issue and follow-up with other issue

sowmyassp, 2022-05-16:
@cjandhyala Its a valid issue, we earlier discussed about different issue 
If you check the log, I have generated the invalid code and parsed the same but commissioning is successful 
[image]

sowmyassp, 2022-05-18:
Its part of version that QR code is invalid

vivien-apple, 2022-05-19:
For the version specific bits, it should be covered by #18606.
That said I suspect that if I close this issue as if once #18606 lands a new one will popup mentioning commissioning flow or the test vendor id :/

sowmyassp, 2022-05-20:
For the version specific bits I am getting "Run command failure: ../../third_party/connectedhomeip/src/controller/SetUpCodePairer.cpp:50: CHIP Error 0x0000002F: Invalid argument
"
Commit used : (d346617f388398b40a207c600fb0c0dee3321902)

Steps used : 
DUT side :
 ./chip-all-clusters-app --version 0 --vendor-id 0xFFF1 --product-id 0x8001 --custom-flow 2 --capabilities 4 --discriminator 3840 --passcode 20202021

TH side : 
./chip-tool pairing code 1 MT:034J029Q00KA0648G00

The command what I used is correct? Or is it a BUG?

yufengwangca, 2022-05-24:
I am not able to reproduce this issue after I synced to ToT :

DUT side :
sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app --wifi (Discriminator - 3840, Passcode - 20202021)
QR code received : MT:-24J0YXE00KA0648G00

TH side
1../chip-tool payload generate-qrcode --discriminator 3840 --setup-pin-code 20202021 --version 1 --vendor-id 65521 --product-id 0x8001 --commissioning-mode 2 --rendezvous 4

QR Code: MT:.24J029Q00KA0648G00

./chip-tool pairing qrcode 12344321 MT:.24J029Q00KA0648G00

The pairing process went through successfully. 

If I use incorrect QR code, such as:
./chip-tool payload generate-qrcode --discriminator 3844 --setup-pin-code 20202021 --version 1 --vendor-id 65521 --product-id 0x8001 --commissioning-mode 2 --rendezvous 4
CHIP:TOO: QR Code: MT:.24J0YJL14KA0648G00

The pairing process failed with the following error:
......
[1653427304.574983][16505:16505] CHIP:CTL: Couldn't get ExampleOpCredsCAKey from storage: ../../../examples/chip-tool/config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1653427304.577321][16505:16505] CHIP:CTL: Couldn't get ExampleOpCredsICAKey from storage: ../../../examples/chip-tool/config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage

[1653427304.608822][16505:16511] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1653427304.608838][16505:16511] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1653427305.609527][16505:16511] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1653427305.609553][16505:16511] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1653427307.610055][16505:16511] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1653427307.610081][16505:16511] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1653427311.611398][16505:16511] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1653427311.611429][16505:16511] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.

liamgonyea, 2022-05-25:
@yufengwangca I noticed in your testing here that the invalid QR code needs to have the same discriminator as @sowmyassp's testing (3840) so that you run into the issue as well as supply an invalid `version`=2.

It sounds like Vivian fixed this but to use **Version** bit string as an example:
For `TC-DD-3.14` Step 2a/b, All parts of the QR code should be the same as `chip-all-clusters-app`'s default configuration (as in no arguments supplied when running all-clusters-app) except for the `version`, this should be invalid one (i.e. '010' or any non-zero 3-bit value) as done above [here](https://github.com/project-chip/connectedhomeip/issues/17205#issuecomment-1127249024)

@sowmyassp This testing command mentioned [here](https://github.com/project-chip/connectedhomeip/issues/17205#issuecomment-1132815427) should be more like `./chip-all-clusters-app --version 2` if you want to test the invalid `version` string
> * DUT side :
>   ./chip-all-clusters-app --version 0 --vendor-id 0xFFF1 --product-id 0x8001 --custom-flow 2 --capabilities 4 --discriminator 3840 --passcode 20202021

yufengwangca, 2022-05-26:
@liamgonyea 

I am not able to reproduce this issue on the latest ToT, which modify the pairing command format. From the current code, the version need to be 0 to pass the pairing

bool PayloadContents::CheckPayloadCommonConstraints() const
{
    // A version not equal to 0 would be invalid for v1 and would indicate new format (e.g. version 2)
    if (version != 0)
    {
        return false;
    }
.....
}

Here is the commands and result of three pairing scenarios:

1. With correct QR code, the pairing went through successfully. 
```
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool payload generate-qrcode --discriminator 3840 --setup-pin-code 20202021 --version 0 --vendor-id 65521 --product-id 0x8001 --commissioning-mode 2 --rendezvous 4
[1653522886.366049][233306:233306] CHIP:TOO: QR Code: MT:-24J029Q00KA0648G00
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ 
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool pairing code 12344321 MT:-24J029Q00KA0648G00
....
[1653523005.014281][234019:234024] CHIP:CTL: Successfully finished commissioning step 'SendComplete'
[1653523005.014294][234019:234024] CHIP:CTL: Commissioning stage next step: 'SendComplete' -> 'Cleanup'
[1653523005.014307][234019:234024] CHIP:CTL: Performing next commissioning step 'Cleanup'
[1653523005.014347][234019:234024] CHIP:CTL: Successfully finished commissioning step 'Cleanup'
[1653523005.014362][234019:234024] CHIP:TOO: Device commissioning completed with success
[1653523005.014406][234019:234024] CHIP:DMG: ICR moving to [AwaitingDe]
[1653523005.014422][234019:234024] CHIP:EM: Sending Standalone Ack for MessageCounter:4649462 on exchange 8003i
[1653523005.014454][234019:234024] CHIP:IN: Prepared secure message 0x7fe42bffdbe8 to 0x0000000000BC5C01 (1)  of type 0x10 and protocolId (0, 0) on exchange 8003i with MessageCounter:8026439.
[1653523005.014475][234019:234024] CHIP:IN: Sending encrypted msg 0x7fe42bffdbe8 with MessageCounter:8026439 to 0x0000000000BC5C01 (1) at monotonic time: 0000000005CC8065 msec
[1653523005.014533][234019:234024] CHIP:EM: Flushed pending ack for MessageCounter:4649462 on exchange 8003i
[1653523005.014773][234019:234019] CHIP:CTL: Shutting down the commissioner
[1653523005.014842][234019:234019] CHIP:CTL: Shutting down the controller
[1653523005.014862][234019:234019] CHIP:CTL: Shutting down the commissioner
[1653523005.014895][234019:234019] CHIP:CTL: Shutting down the controller
[1653523005.014914][234019:234019] CHIP:IN: Expiring all connections for fabric 1!!
[1653523005.014927][234019:234019] CHIP:IN: SecureSession MarkForRemoval 0x559aa737d1a8 Type:2 LSID:51359
[1653523005.014942][234019:234019] CHIP:IN: SecureSession Released 0x559aa737d1a8 Type:2 LSID:51359
[1653523005.014971][234019:234019] CHIP:CTL: Shutting down the commissioner
[1653523005.015005][234019:234019] CHIP:CTL: Shutting down the controller
[1653523005.015017][234019:234019] CHIP:IN: Expiring all connections for fabric 2!!
[1653523005.015037][234019:234019] CHIP:CTL: Shutting down the commissioner
[1653523005.015068][234019:234019] CHIP:CTL: Shutting down the controller
[1653523005.015080][234019:234019] CHIP:IN: Expiring all connections for fabric 3!!
[1653523005.015098][234019:234019] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1653523005.015427][234019:234019] CHIP:DMG: IM WH moving to [Uninitialized]
[1653523005.015441][234019:234019] CHIP:DMG: IM WH moving to [Uninitialized]
[1653523005.015452][234019:234019] CHIP:DMG: IM WH moving to [Uninitialized]
[1653523005.015463][234019:234019] CHIP:DMG: IM WH moving to [Uninitialized]
[1653523005.015475][234019:234019] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1653523005.015509][234019:234019] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1653523005.015637][234019:234019] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-zVeofK)
[1653523005.015905][234019:234019] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1653523005.015938][234019:234019] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1653523005.015957][234019:234019] CHIP:DL: Inet Layer shutdown
[1653523005.015971][234019:234019] CHIP:DL: BLE shutdown
[1653523005.015983][234019:234019] CHIP:DL: System Layer shutdown
[1653523005.016017][234019:234019] CHIP:IN: SecureSession Released 0x559aa737d0c0 Type:1 LSID:51358
```
2. With incorrect password, the pairing failed immediately. 
```
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool payload generate-qrcode --discriminator 3840 --setup-pin-code 20201111 --version 0 --vendor-id 65521 --product-id 0x8001 --commissioning-mode 2 --rendezvous 4
[1653523021.811270][234305:234305] CHIP:TOO: QR Code: MT:-24J029Q00INJZ38G00
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ 
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ 
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ 
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool pairing code 12344321 MT:-24J029Q00INJZ38G00
......
[... 159 lines cut ...]
[1653523105.197536][234479:234484] CHIP:DIS: MINMDNS:     Commission SRV for instance 17ECD0C0B8FE47D1: F4939FF4A4DA0000.local
[1653523105.197554][234479:234484] CHIP:DIS: MINMDNS: received PTR record for _CM._sub._matterc._udp.local
[1653523105.197568][234479:234484] CHIP:DIS: MINMDNS: received AAAA record for F4939FF4A4DA0000.local
[... cut ...]
[1653523105.198943][234479:234484] CHIP:DIS: MINMDNS: received TXT record for 17ECD0C0B8FE47D1._matterc._udp.local
[1653523106.188393][234479:234484] CHIP:DIS: MINMDNS: Sending query ANY/MULTICAST for _CM._sub._matterc._udp.local
[1653523106.188754][234479:234484] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1653523106.188774][234479:234484] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
```

sowmyassp, 2022-05-26:
> I am not able to reproduce this issue after I synced to ToT :
>
> DUT side : sudo rm -rf /tmp/chip_* sudo ./chip-all-clusters-app --wifi (Discriminator - 3840, Passcode - 20202021) QR code received : [...]

I used code for invalid version on TOT 
**Command used** :
- DUT :   ./chip-all-clusters-app --version 2 --vendor-id 0xFFF1 --product-id 0x8001 --custom-flow 2 --capabilities 4 --discriminator 3840 --passcode 20202021
- TH : ./chip-tool pairing code 1 MT:034J029Q00KA0648G00

**The result I got is** 
ERROR: Run command failure: ../../third_party/connectedhomeip/src/controller/SetUpCodePairer.cpp:50: CHIP Error 0x0000002F: Invalid argument

@yufengwangca can you once verify for the invalid version bit and also make sure you pass the valid parameters in the above command i.e., ./chip-all-clusters-app --version 2 --vendor-id 0xFFF1 --product-id 0x8001 --custom-flow 2 --capabilities 4 --discriminator 3840 --passcode 20202021 where only the version parameter should be invalid so that we will get expected outcome

yufengwangca, 2022-05-26:
@sowmyassp 

As I mentioned above, version number other than 0 is invalid in current implementation, you won't generate the QR code, is there say we should ignore the version check?

    // A version not equal to 0 would be invalid for v1 and would indicate new format (e.g. version 2)
    if (version != 0)
    {
        return false;
    }
.....

sankarselvam, 2022-05-27:
@cjandhyala  and @liamgonyea  Can you please clarify YUfeng's question? If required, we can set-up a call and clarify to close this issue.

cjandhyala, 2022-05-27:
@yufengwangca Could you try pairing with the QR code that Sowmya suggested here i.e. MT:034J029Q00KA0648G00 , Whatever the QRcode you are using is different to what Sowmya is using. Your QR code are using invalid descriminator and passcode ,whereas Sowmya's QR code is using all valid discriminator, passcode ..etc but not version field. With her QR code she is seeing the device is provisioned, which it should not.

yufengwangca, 2022-05-31:
No, I am not able to pass with Sowmya suggested QR code.

```
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool pairing code 12344321 MT:034J029Q00KA0648G00
[1654009135.425264][1332144:1332144] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1654009135.425715][1332144:1332144] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1654009135.425799][1332144:1332144] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1654009135.425863][1332144:1332144] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1654009135.426115][1332144:1332144] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-68HyvQ)
[1654009135.426426][1332144:1332144] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1654009135.426461][1332144:1332144] CHIP:DL: NVS set: chip-counters/reboot-count = 2 (0x2)
[1654009135.427075][1332144:1332144] CHIP:DL: Got Ethernet interface: eno1
[1654009135.427480][1332144:1332144] CHIP:DL: Found the primary Ethernet interface:eno1
[1654009135.428152][1332144:1332144] CHIP:DL: Failed to get WiFi interface
[1654009135.428166][1332144:1332144] CHIP:DL: Failed to reset WiFi statistic counts
[1654009135.428214][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=0
[1654009135.428297][1332144:1332144] CHIP:IN: UDP::Init bound to port=43242
[1654009135.428313][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=0
[1654009135.428360][1332144:1332144] CHIP:IN: UDP::Init bound to port=53446
[1654009135.428374][1332144:1332144] CHIP:IN: BLEBase::Init - setting/overriding transport
[1654009135.428386][1332144:1332144] CHIP:IN: TransportMgr initialized
[1654009135.428410][1332144:1332144] CHIP:DIS: Init fabric pairing table with server storage
[1654009135.434042][1332144:1332144] CHIP:ZCL: Using ZAP configuration...
[1654009135.436030][1332144:1332144] CHIP:CTL: System State Initialized...
[1654009135.436081][1332144:1332144] CHIP:CTL: Setting attestation nonce to random value
[1654009135.436108][1332144:1332144] CHIP:CTL: Setting CSR nonce to random value
[1654009135.436143][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=5550
[1654009135.436185][1332144:1332144] CHIP:IN: UDP::Init bound to port=5550
[1654009135.436198][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=5550
[1654009135.436235][1332144:1332144] CHIP:IN: UDP::Init bound to port=5550
[1654009135.436246][1332144:1332144] CHIP:IN: TransportMgr initialized
[1654009135.436316][1332144:1332144] CHIP:CTL: Setting attestation nonce to random value
[1654009135.436342][1332144:1332144] CHIP:CTL: Setting CSR nonce to random value
[1654009135.436567][1332144:1332144] CHIP:CTL: Couldn't get ExampleOpCredsCAKey from storage: ../../../examples/chip-tool/config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1654009135.438810][1332144:1332144] CHIP:CTL: Couldn't get ExampleOpCredsICAKey from storage: ../../../examples/chip-tool/config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storage
[1654009135.439439][1332144:1332144] CHIP:CTL: Generating RCAC
[1654009135.439931][1332144:1332144] CHIP:CTL: Generating ICAC
[1654009135.440366][1332144:1332144] CHIP:CTL: Generating NOC
[1654009135.440967][1332144:1332144] CHIP:DIS: Verifying the received credentials
[1654009135.442183][1332144:1332144] CHIP:DIS: Added new fabric at index: 0x1, Initialized: 1
[1654009135.442202][1332144:1332144] CHIP:DIS: Assigned compressed fabric ID: 0x2F13D40C1BDFCDF2, node ID: 0x000000000001B669
[1654009135.443456][1332144:1332144] CHIP:DIS: Fabric (0x1) persisted to storage. Calling OnFabricPersistedToStorage
[1654009135.443721][1332144:1332144] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x2F13D40C1BDFCDF2
[1654009135.443754][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=5550
[1654009135.443806][1332144:1332144] CHIP:IN: UDP::Init bound to port=5550
[1654009135.443821][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=5550
[1654009135.443861][1332144:1332144] CHIP:IN: UDP::Init bound to port=5550
[1654009135.443874][1332144:1332144] CHIP:IN: TransportMgr initialized
[1654009135.443918][1332144:1332144] CHIP:CTL: Setting attestation nonce to random value
[1654009135.443947][1332144:1332144] CHIP:CTL: Setting CSR nonce to random value
[1654009135.444735][1332144:1332144] CHIP:CTL: Generating NOC
[1654009135.445268][1332144:1332144] CHIP:DIS: Verifying the received credentials
[1654009135.446118][1332144:1332144] CHIP:DIS: Added new fabric at index: 0x2, Initialized: 1
[1654009135.446131][1332144:1332144] CHIP:DIS: Assigned compressed fabric ID: 0x79394E1ECC27D77B, node ID: 0x000000000001B669
[1654009135.447107][1332144:1332144] CHIP:DIS: Fabric (0x2) persisted to storage. Calling OnFabricPersistedToStorage
[1654009135.447306][1332144:1332144] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x79394E1ECC27D77B
[1654009135.447329][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=5550
[1654009135.447365][1332144:1332144] CHIP:IN: UDP::Init bound to port=5550
[1654009135.447375][1332144:1332144] CHIP:IN: UDP::Init bind&listen port=5550
[1654009135.447402][1332144:1332144] CHIP:IN: UDP::Init bound to port=5550
[1654009135.447412][1332144:1332144] CHIP:IN: TransportMgr initialized
[1654009135.447449][1332144:1332144] CHIP:CTL: Setting attestation nonce to random value
[... cut ...]
[1654009135.467944][1332144:1332144] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1654009135.467954][1332144:1332144] CHIP:DL: Inet Layer shutdown
[1654009135.467960][1332144:1332144] CHIP:DL: BLE shutdown
[1654009135.467964][1332144:1332144] CHIP:DL: System Layer shutdown
[1654009135.467990][1332144:1332144] CHIP:TOO: Run command failure: ../../../examples/chip-tool/third_party/connectedhomeip/src/controller/SetUpCodePairer.cpp:50: CHIP Error 0x0000002F: Invalid argument
```

yufengwangca, 2022-05-31:
I think there are something wrong with Sowmya's QR code, since we should not able to generate this code based on our current version check logic in our code. I tried to generate the QR code with only incorrect version info using ToT, and it failed as I expected. 

yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool payload generate-qrcode --discriminator 3840 --setup-pin-code 20202021 --version 2 --vendor-id 65521 --product-id 0x8001 --commissioning-mode 2 --rendezvous 4
[1654009064.237092][1332028:1332028] CHIP:TOO: Run command failure: ../../../examples/chip-tool/third_party/connectedhomeip/src/setup_payload/QRCodeSetupPayloadGenerator.cpp:218: Error 0x0000002F
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$

sowmyassp, 2022-05-31:
- For Custom Flow field bit set to 1 in step 3a and set to 2 in step 3c of test-case TC-DD-3.14  where we have QR code generated for it is  "MT:-24J029Q00KA0648G00" where the  DUT should terminate the commissioning process, but DUT successfully completed the commissioning process

- TC-DD-3.17
1. For VENDOR_ID from the list: 0xFFF1, 0xFFF2, 0xFFF3, 0xFFF4 the manual code (749701123365521327694, 749701123365522327692, 749701123365523327697, 749701123365524327693) the commissioning flow is passing successfully
2. For invalid PRODUCT_ID the manual code (749701123365521000006)  the commissioning flow is passing successfully

somu1710, 2022-06-01:
> OK, there is clearly discrepancy between the test plan and current implementation, as I said, we need to clarify if version other than 0 is valid

@liamgonyea Do you have any comment on this

sowmyassp, 2022-06-01:
> OK, there is clearly discrepancy between the test plan and current implementation, as I said, we need to clarify if version other than 0 is valid

@liamgonyea would you like to comment on this

liamgonyea, 2022-06-07:
> > OK, there is clearly discrepancy between the test plan and current implementation, as I said, we need to clarify if version other than 0 is valid
> 
> @liamgonyea would you like to comment on this

There are discrepancies between the test plan and the current implementation. 
@sowmyassp has laid out the known discrepancies above [here](https://github.com/project-chip/connectedhomeip/issues/17205#issuecomment-1142391858): 

> * For Custom Flow field bit set to 1 in step 3a and set to 2 in step 3c of test-case TC-DD-3.14  where we have QR code generated for it is  "MT:-24J029Q00KA0648G00" where the  DUT should terminate [...]

#### Note, that for each of these discrepancies the pairing flow is commissioning a device successfully _*but*, the commissioning should not be successful._ Because of this successful pairing with invalid QR codes, the current implementation is Failing these test cases:
**TC-DD-3.14**

- `Step 3.b` Invalid QR code contains `Custom Flow=1` (user-intent flow) but device uses `Custom Flow=2` (custom commissioning flow)
- `Step 3.d` Invalid QR code contains `Custom Flow=2` (custom commissioning flow) but device uses `Custom Flow=0` (standard flow)

**TC-DD-3.17** 

- `Step 6.b` Vendor ID equal to 0xFFF1, 0xFFF2, 0xFFF3, 0xFFF4 in the invalid manual code
- `Step 7.b` Product ID equal to 0x0000 in the invalid manual code

yufengwangca, 2022-06-07:
OK, the current implementation only prevent the chip-tool from generating the QR code if "version != 0", we should also make change to make the chip-tool fail to commission with the QR code generated in other way with  "version != 0".

bzbarsky-apple, 2022-06-10:
> we should also make change to make the chip-tool fail to commission with the QR code generated in other way with "version != 0"

That's already the case.  The comments above are not about versions.

Just to reiterate what I said in #18272:

* TC-DD-3.14 steps 3.b and 3.d probably need a new, clean issue that clearly describes the problem and points to the relevant spec sections that require the commissioner to have that behavior.
* TC-DD-3.17 step 6.b is not reasonable.  Commissioning test vendor ids, at the commissioner's discretion must be allowed.  Where does the spec say that should not be allowed?  And certainly our _test tools_ need to be able to commision those vendor ids.
* TC-DD-3.17 step 7.b worked for me correctly (refused to commission) last I checked.  If there is still a problem here, please file a new issue, with the exact SHA used and the exact steps to reproduce.

@liamgonyea @sowmyassp Please file clean issues on the remaining bits here as needed, one issue per problem instead of lumping them all together.

sowmyassp, 2022-06-14:
Sure @bzbarsky-apple I will file ticket on remaining bits and link it here.

closed 2022-06-16 by sowmyassp

sowmyassp, 2022-06-16:
[19676](https://github.com/project-chip/connectedhomeip/issues/19676)
[19680](https://github.com/project-chip/connectedhomeip/issues/19680)

andy31415, 2022-06-17:
@sowmyassp  please provide the spec reference for the requirements of the test plan.

I believe our current SDK implementation is "If a QR code is provided, always use the long discriminator to commission". I believe this also means that we ignore product/vendorid and such and this seems like ok to me unless the spec explicitly forbids it.

I agree that it could be made more robust (i.e. I scan a QR code that has all fields, then I discover a device and can validate that these IDs match or not to not mistakenly commission something else) however I want to ensure that the spec says I MUST do this (aka SHALL wording). If the spec does not specify this, this is not a spec requirement.

andy31415, 2022-06-17:
Specific things that I consider bugs at this point (and I believe were fixed):
  - we should reject invalid version for the QR code (because if version differs, it means we should not know how to correctly parse it if the format changed)
  - Good long discriminator should pair, bad long discriminator should not (because discriminator is the only thing guaranteed to be present in the QR code and advertisement data ... vendor and product IDs are not guaranteed)
 
 I do NOT believe that not using or validating vendorid/productid is a spec requirement, so even though nice I don't think they should apply to a "Spec blocker" bug. We can open separate bugs for such enhancements, likely they will not make it to 1.0 and will not be tagged spec.

yufengwangca, 2022-06-17:
Currently, we only use discriminator for device discovery, both long and short depending on which one is used in the QR code

```
    currentFilter.type = payload.isShortDiscriminator ? Dnssd::DiscoveryFilterType::kShortDiscriminator
                                                      : Dnssd::DiscoveryFilterType::kLongDiscriminator;
    currentFilter.code =
        payload.isShortDiscriminator ? static_cast<uint16_t>((payload.discriminator >> 8) & 0x0F) : payload.discriminator;
```

We shall not use any optional fields for device filtering, we need to correct relative test plans

yufengwangca, 2022-06-18:
Per sync with @andy31415 and @bzbarsky-apple, I think our current SDK implementation align with the current spec.

With current implementation, commissioner always use the given discriminator for device discovery, after device is discovered, commissioner ignore the ones with CM=0, since those are not listening for PASE. For the ones with CM != 0, try PASE.