project-chip/connectedhomeip #17206: Discriminator value doesn't match with DUT and TH

sowmyassp, 2022-04-08:
Issue : 
While launching the DUT I passed ./chip-all-clusters-app with discriminator 3841 but passing parse-setup-payload with manual code generated in DUT on TH I am getting discriminator as 3840.

Steps to reproduce : 

1. sudo ./chip-all-clusters-app --wifi --discriminator 3841
[image]
2../chip-tool payload parse-setup-payload 34970112332
[image]

tcarmelveilleux, 2022-04-08:
@sowmyassp Can you please provide textual logs showing the entire logs leading to setup payload being given to you when you run `./chip-all-clusters-app` ?

sowmyassp, 2022-04-12:
@tcarmelveilleux 
[log_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8472733/log_cntrl.txt)
[log_dut.txt](https://github.com/project-chip/connectedhomeip/files/8472734/log_dut.txt)

erjiaqing, 2022-04-27:
Seems this is a expected behavior

Spec: 5.1.4, table 37

DISCRIMINATOR 4 Yes 4 Most-significant bits of the 12-bits Discriminator described above

sowmyassp, 2022-04-27:
Yes the DISCRIMINATOR value should be 4 bits as per the spec but I  have launched the all-clusters-app with discriminator 3841 so the same should be displayed on TH once the commission is done right?

erjiaqing, 2022-04-28:
> Yes the DISCRIMINATOR value should be 4 bits as per the spec but I have launched the all-clusters-app with discriminator 3841 so the same should be displayed on TH once the commission is done right?

Since you are parsing the setup payload instead of reading the discriminator from the DUT, it really has little to do with the discriminator.

I would suggest it show something like "discriminator: 0xFxx", or "discriminator: 3840-4095" instead of a single concrete number.

sowmyassp, 2022-04-28:
> > Yes the DISCRIMINATOR value should be 4 bits as per the spec but I have launched the all-clusters-app with discriminator 3841 so the same should be displayed on TH once the commission is done right?
> 
> Since you are parsing the setup payload instead of reading the discriminator from the DUT, it really has little to do with the discriminator.
> 
> I would suggest it show something like "`discriminator: 0xF??`", or "`discriminator: 3840-4095`" instead of a single concrete number.

erjiaqing, 2022-05-09:
> > > Yes the DISCRIMINATOR value should be 4 bits as per the spec but I have launched the all-clusters-app with discriminator 3841 so the same should be displayed on TH once the commission is done right?
> > 
> > 
> > Since you are parsing the setup payload instead of reading the discriminator from the DUT, it really has little to do with the discriminator.
> > I would suggest it show something like "`discriminator: 0xF??`", or "`discriminator: 3840-4095`" instead of a single concrete number.
> 
> Ok then we don't have any issues here right?

Yes, there should be a UI / UX improvement instead of a bug.

closed 2022-05-10 by sowmyassp