project-chip/connectedhomeip #17208: AddTrustedRootCertificate is missing validations from spec

tcarmelveilleux, 2022-04-08:
#### Problem
The following TODOs exists in AddTrustedRootCertificate handler:

```
    // TODO: Ensure we do not duplicate roots in storage, and detect "same key, different cert" errors
    // TODO: Validate cert signature prior to setting.
    VerifyOrExit(gFabricBeingCommissioned.SetRootCert(RootCertificate) == CHIP_NO_ERROR, status = EMBER_ZCL_STATUS_INVALID_FIELD);
```

Both the duplicate check and signature validation are required. Furthermore, the assumption that the root should be stored in the "gFabricBeingCommissioned" means we cannot reuse roots for multiple clients without re-adding the root (which is not required by spec)

 #### Proposed Solution

- Implement signature verification check of root
- Implement validity checks required by spec on AddTrustedRootCertificate

woody-apple, 2022-07-22:
Spec Issue Review: Assigning @bzbarsky-apple to confirm if there's any remaining work, and close if not.

emargolis, 2022-07-22:
The first TODO became obsolete and was removed in #21060
The second TODO was addressed in #20562 
This ticket can be closed

bzbarsky-apple, 2022-07-23:
Yep, that matches my understanding.  Thank you @emargolis !

closed 2022-07-23 by bzbarsky-apple